felis/UHS30 #1: Development Environment Advice / Recomendation

marcah, 2015-09-16:
I acknowledge that there is no technical support available for UHS30 at this time.

I have indeed committed to the use of UHS30 for a project.  Thus far, I have been utilizing the UHS_FS_NEW_DEMO to test the mounting and access to a thumb drive.  The test seems to run for days suggesting that the current UHS30 code is reliable.

I am just about to begin developing the real application which will include a few sensors and two USB thumb drives and a SD memory card.  The SD card is intended to be the primary data storage location with the two USB devices available for a user to periodically swap.

My use of UHS30 thus far has been via Arduino IDE version 1.6.5.  I have as well successfully built the test applications using the Visual Micro plugin for Atmel Studio 6.2.  While both environments seem to work, my desire is to shift development to Atmel Studio 6.2 w/o the use of the Visual Studio plugin.
I have created a core library from the Arduino 1.6.5 IDE and libraries for all of the non UHS30 Arduino libraries that are necessary when using the IDE or Visual Micro.  I seem to be having some difficulty in creating a standalone UHS30 library within Visual Studio.

I am looking for some advice or recommendations regarding my plan to shift to the use of Visual Studio.

My current strategy in attempting to create a UHS30 library is actually to create individual libraries for each of the directories within the UHS30\library structure.  This may indeed be problematic.

Any thoughts, feedback or advice would be greatly appreciated.

Regards,
-marc

Lauszus, 2015-09-23:
@marcah sorry for the long delay. I never use Visual Studio, but just use Eclipse. I would really recommend the Arduino Eclipse plugin: http://eclipse.baeyens.it/ which I use for larger projects.

xxxajk, 2015-09-24:
Note also that normally I do not use the Arduino IDE either. I use Netbeans
and GNU make on linux. Also, Netbeans does not require any plugins, it just
works.

I guess the point here is to use what ever works for you. I suggest trying
everything, and decide on your own.

marcah, 2015-09-26:
Thanks for the responses regarding development tools... I really appreciate your feedback.

I have been utilizing the UHS_FS_NEW_DEMO sketch as the basis for some experimentation.  Using Arduino IDE version 1.6.5, the demo works as expected with no issues.  My application also requires the use of SD.  I have managed to "hack" together a test that combines UHS30 & SD that works both via the Arduino IDE and via Visual Studio using the Visual Micro plugin.  While the initial tests are functional, I am not real comfortable with the elements of Visual Micro that I don't understand.  There are dependencies in the build environment on Visual Studio, Visual Micro and the Arduino IDE.  It feels like it could require a great deal of "luck" to duplicate the development environment on another system.

Here is my real question...  For those of you working with UHS30, are there examples / demos that work with in an non Arduino IDE environment?  In trying to get the UHS_FS_NEW_DEMO to build with Visual Studio (w/o Visual Micro & Arduino IDE), I suspect that the current Arduino IDE preprocessor issues preclude the demo from properly building on a more 'pure' development environment such as in use by Lauszus & xxxajk.

Any thoughts / advice would be appreciated.

BTW...  The existing UHS_FS_NEW_DEMO seems to be demonstrating that UHS30 is indeed quite solid.  i have significant functionality yet to develop, but, it seems that UHS30 is the right path.

Regards,
-marc

marcah, 2015-09-26:
Didn't intend to close...
-m

xxxajk, 2015-09-26:
Only one other and afaik, he uses the ide.
Very happy that it is working out for you.
I realize there isn't any documentation for all the available functions for
the file system, but I'm sure you have looked at the headers and found out
what is available.
UHS 3 will be getting more expansions very soon, as the current contract I
am working on is going to be requiring it for ARM mcu. Whats cool is that
the same will apply to the host shield automatically, because of how
flexible the new core API is.
As I said before, use whatever works for you.

Lauszus, 2015-09-30:
@marcah note the preprocessor issues with the Arduino IDE will be over in the next version of the IDE, as they are now using a new builder: https://github.com/arduino/arduino-builder.

xxxajk, 2015-10-01:
Not that it matters, since UHS 3 is all loaded inline via headers ;-)

marcah, 2015-10-01:
Thanks to both of you for the feedback.

Actually, I wonder if in fact the new builder will help.  One of the challenges with the current preprocessor is the requirement within the Arduino IDE to reference (include) all .h files in the main sketch.  In the case of loading inline, reference to the same .h files in other files results in an error related to loading via a “driver.”  I also see this same behavior in using Atmel Studio with the Visual Micro tools as it relies upon the Arduino IDE to maintain sketch compatibility.  My hope had been to utilize Atmel Studio w/o the Visual Micro tools but thus far I have not been able to get UHS30 to build properly w/o using Visual Micro.

In any case, I suspect that the new builder is going to be a great addition.  However, I suspect along the way there will be some challenges with existing libraries and some sketches that currently have some structure dependent on the current builder limitations.

Is there an easy means to temporarily disable loading inline that one could experiment with?

Thanks for taking the time to share your thoughts.

-m

xxxajk, 2015-10-01:
I am unaware of any issues with inlining the code.

Are you using multiple C++ source files? If so, just include the header
without doing any of the defines and it will work.

marcah, 2015-10-01:
Thanks again for the advice.

I am indeed using multiple C++ source files.  I will test your suggestion of not including the defines and let you know how it goes.

BTW… while my UHS30 tests are still pretty simple, seems to be very reliable.  There are times with new components that is sort of feels like a little “luck” is required… not so in this case.  I appreciate your work and that of the others that are making my developments possible.

Regards,
-m

marcah, 2015-10-02:
Hey Andrew,

I have performed some testing this afternoon based on the “UHS_FS_NEW_DEMO”.  My objective is to try your suggestion to "just include the header without doing any of the defines."

For the purpose of my experiments, I have created a two file test: 1) UHS_FS_NEW_DEMO_20151001.ino, 2) uhs30.cpp .

In the first example, uhs30.cpp is essentially empty; UHS_FS_NEW_DEMO_20151001.ino is as follows:

#define LOAD_USB_HOST_SHIELD

#include <Wire.h>
#include <SPI.h>

#include <RTClib.h>
#include <UHS_host.h>
#include <USB_HOST_SHIELD.h>
#include <UHS_HUB.h>
#include <UHS_BULK_STORAGE.h>
#include <UHS_FS.h>

void setup()
{
}

void loop()
{
}

With the #define in place, the test builds without errors.

Leaving out the #define results in the following error:
In file included from UHS_FS_NEW_DEMO_20151001.ino:8:0: C:\Users\marc.KLAHANIE\Documents\Arduino\libraries\USB_HOST_SHIELD/USB_HOST_SHIELD.h:325:2: error: #error "define LOAD_USB_HOST_SHIELD in your sketch, never include USB_HOST_SHIELD.h in a driver."
 #error "define LOAD_USB_HOST_SHIELD in your sketch, never include USB_HOST_SHIELD.h in a driver."
  ^
Error compiling.

In the second example, UHS_FS_NEW_DEMO_20151001.ino is configured as above, including the #define.
The file uhs30.cpp is configured as follows:

#define LOAD_USB_HOST_SYSTEM
//#define LOAD_USB_HOST_SHIELD
#define LOAD_UHS_BULK_STORAGE
#define LOAD_RTCLIB
#define LOAD_GENERIC_STORAGE
#define _USE_MAX3421E_HOST 1

#include <Wire.h>
#include <SPI.h>

#include <RTClib.h>
#include <UHS_host.h>
#include <USB_HOST_SHIELD.h>
#include <UHS_HUB.h>
#include <UHS_BULK_STORAGE.h>
#include <UHS_FS.h>

With the “#define LOAD_USB_HOST_SHIELD” commented out, the same error as above is reported.

With the “#define LOAD_USB_HOST_SHIELD” included, there are many reports of multiply defined symbols.

I may have misunderstood your suggestion to exclude all of the #define statements from the .ino sketch.  It does appear that the "#define LOAD_USB_HOST_SHIELD" is required in the main .ino but in the case of the second .cpp file both including and not including the "#define LOAD_USB_HOST_SHIELD" results in errors.

I should have mentioned before that all of the above test cases were run using Arduino IDE 1.6.5.  My default development environment is currently Atmel Studio with the Visual Micro addin.  In any case, I suspected that perhaps the current Arduino IDE builder could be causing an issue.  Can you confirm that in the case of your default development environment, the "just include the header without doing any of the defines" works for you?

Again, thanks for your advice and your work toward providing UHS30 for the community!
Regards,
-marc

xxxajk, 2015-10-02:
Indeed you did misunderstand :-)
Worse, this needs to be documented better.
It would be very helpful if you could directly email me your actual project, so that I can assist you.
It is very hard to determine what the problem is, because you are mixing concepts, but here is the basic way to get what you want if I am correct in my assumption of what you are attempting to do.

First, in your main sketch (ok, lets be realistic here, .ino is nothing but a cpp file) is where you have all your settings and includes for everything you will be doing.
To access these from another cpp file within your project, you only include the parts you need.
In other words, lets say you got a cpp file that is going to be doing file access. All you do is include the UHS_FS.h file.

marcah, 2015-10-02:
Hey Andrew,
Thanks for your continued advice and willingness to clarify.

I suspect that in general, we are “on the same page” in terms of how things work & should work.  The one point that we may not be fully aligned on is as follows:

1)      Indeed the .ino must include all of the .h includes that will be used in any of the additional .cpp source files (this is really a requirement of the existing Arduino IDE builder which as Lauszus pointed out will be fixed in a future release).

2)      It appears that “#define LOAD_USB_HOST_SHIELD” must appear before “#include <USB_HOST_SHIELD.h>” in the .ino or the following results:    #error "define LOAD_USB_HOST_SHIELD in your sketch, never include USB_HOST_SHIELD.h in a driver."

3)      It appears that “#define LOAD_USB_HOST_SHIELD” must appear before “#include <USB_HOST_SHIELD.h>” in any additional .cpp source files or the same error is produced:    #error "define LOAD_USB_HOST_SHIELD in your sketch, never include USB_HOST_SHIELD.h in a driver."

4)      Here is the issue…  It appears that when “#define LOAD_USB_HOST_SHIELD” appears in a second .cpp source file, the inline a second time results in multiply defined symbols, as would be expected.

5)      Question… Should it be possible to “#include <USB_HOST_SHIELD.h>” w/o the #define resulting in no “inline” of the code?  At least w/ Arduino IDE 1.6.5, this doesn’t appear to be possible.

My conclusions…

1)      With Arduino IDE 1.6.5, all UHS30 functionality depending on USB_HOST_SHIELD.h must be contained within the .ino and cannot be located in another .cpp source file.

2)      Potentially, with Arduino IDE 1.6.6 including the new builder, all includes will possibly not be required to be specified in the .ino, allowing for the placement of UHS30 functionality in another .cpp source file.

3)      I imagine that even with Arduino IDE 1.6.6, it may not be possible to include USB_HOST_SHIELD.h in multiple .cpp source files due to what appears to be a current requirement to precede the #include with the associated #define.

Summary

1)      UHS30 is working wonderfully for me at this point.

2)      I suspect there is an issue related to the inline logic.

Best Regards,
-marc

xxxajk, 2018-02-16:
Closing issue because no more traffic. Reopen if you still misunderstand.

closed 2018-02-16 by xxxajk