felis/UHS30 #4: Kinetis Reset AKA Detach

felis, 2015-10-13:
Possible bug:
https://github.com/felis/UHS30/blob/master/libraries/UHS_KINETIS_FS_HOST/UHS_KINETIS_FS_HOST_INLINE.h#L348

In PIC32 which is likely based on the same SIE this bit means 'reset' in device mode and 'detach' in host mode (that's our mode of interest). Xxxajk - please confirm or deny.

xxxajk, 2015-10-13:
condet = true; <--- connection/disconnection is detected.
https://github.com/felis/UHS30/blob/master/libraries/UHS_KINETIS_FS_HOST/UHS_KINETIS_FS_HOST_INLINE.h#L350
So yes, same bit function, but on Kinetis it also means attach.

xxxajk, 2015-10-13:
To be clear, I mean 'condet' not the bit. condet just flags to say there is a line change and to check line status.

xxxajk, 2018-02-16:
Reopen if you have additional questions...

closed 2018-02-16 by xxxajk