felis/UHS30 #7: interrupts(), noInterrupts() - new home?

felis, 2015-10-14:
ATM they come from Arduino tree and apart from hw-specific code they are called from the core as well as swi (see grep output below). PIC32 won't be able to use it. What would be the best course of action - abstract it out somewhere else, perhaps?

felis@unstable:~/NetBeansProjects/UHS30$ grep -r noInterrupts .
./libraries/UHS_KINETIS_FS_HOST/UHS_KINETIS_FS_HOST_INLINE.h:                noInterrupts();
./libraries/UHS_KINETIS_FS_HOST/UHS_KINETIS_FS_HOST_INLINE.h:                noInterrupts();
./libraries/UHS_KINETIS_FS_HOST/UHS_KINETIS_FS_HOST_INLINE.h:                noInterrupts();
./libraries/UHS_host/mlock.c:        noInterrupts();
./libraries/UHS_host/UHS_usbhost.h:                noInterrupts();
./libraries/UHS_host/UHS_usbhost.h:                noInterrupts();
./libraries/USB_HOST_SHIELD/USB_HOST_SHIELD_INLINE.h:        noInterrupts();
./libraries/USB_HOST_SHIELD/USB_HOST_SHIELD_INLINE.h:        noInterrupts();
./libraries/USB_HOST_SHIELD/USB_HOST_SHIELD_INLINE.h:                noInterrupts();
./libraries/USB_HOST_SHIELD/USB_HOST_SHIELD_INLINE.h:                noInterrupts();
./libraries/dyn_SWI/SWI_INLINE.h:        noInterrupts();
./libraries/dyn_SWI/SWI_INLINE.h:        noInterrupts();
./libraries/dyn_SWI/SWI_INLINE.h:        noInterrupts();
./libraries/dyn_SWI/SWI_INLINE.h:        noInterrupts();
./libraries/UHS_PIC32/UHS_PIC32_INLINE.h:                noInterrupts();
./libraries/UHS_PIC32/UHS_PIC32_INLINE.h:                noInterrupts();
./libraries/UHS_PIC32/UHS_PIC32_INLINE.h:                noInterrupts();

felis, 2015-10-15:
added pic32 def -> https://github.com/felis/UHS30/commit/322a04f8d6e73e352133a53a7a9df9f1a29d68d3

xxxajk, 2015-10-15:
> added pic32 def -> 322a04f

You probably should check if they are already defined.

felis, 2015-10-15:
agree

xxxajk, 2016-10-06:
BTW the latest chipkit core does have these designed... just so you know. It may actually be easier to use that, anyway since I have gotten all of my patches accepted.

xxxajk, 2017-01-18:
This all works with current tools.

closed 2017-01-18 by xxxajk