felis/UHS30 #8: Mount point for USB drive

marcah, 2015-10-15:
I have been successfully experimenting with a number of USB thumb drives all with a blank label.  As expected, the drives mount at "/".  I am now attempting to mount a drive with a non blank label such as 'foo'.  The drive appears to mount but then reports 0 space available on the drive.  With a label of 'foo' on the drive I am passing '/foo' to both 'fs_ready()' and 'fs_getfree()'.  I am using 'UHS_FS_NEW_DEMO' as the basis for my tests.  I recognize that once the reported free space test passes, the following file names will need to include the mount point '/foo' as well.

I suspect that my understanding of the proper mount points is in error.

In anticipation of mounting multiple drives, I have increased _FS_LOCK  from 1 to 4.

Continued thanks for your great work to support UHS30.

xxxajk, 2015-10-15:
_FS_LOCK is for file counts.
Same partition mounts on / fine?

xxxajk, 2015-10-15:
Have you tried "/foo/" instead of "/foo"? 
IIRC (been a while) you may need the trailing forward slash on the path name.

marcah, 2015-10-15:
Thanks for the quick response!

‘foo’  stops with “USB HOST state 60”
‘/foo/’  stops with “USB HOST state 60”
‘foo/’  stops with “USB HOST state 60”

‘/foo’  indicates “USB HOST state 60”, “/foo mounted.”, “Not enough space (fre = 0) to run tests... ”

Removing the label, three different drives all mount fine at ‘/’.

…..

Just noticed that there are two calls to ‘fs_ready()’, the first time the statement “if(mounted != PFAT_VOLUMES) {” fails as mounted == 1 which is the value of PFAT_VOLUMES.

I am using the windows ‘label’ command to set and remove the volume label.  Could I be removing the label ok but not setting the label correctly?  It would seem that the volume itself must be ok as mounting as ‘/’ is fine.

xxxajk, 2015-10-15:
Got a small example code snippit to demo the problem? I'm elevating this to a bug.

marcah, 2015-10-15:
Hello Andrew,
Attached are a couple of log files that show the working (“Root_Works_Log” example mounting “/”) and the failing (“FOO_Fail_Log” example mounting “/foo”).  This test is based on UHS_FS_NEW_DEMO.ino of which I have attached a fragment.  Essentially, I have enhanced a couple of print statements and simply defined a constant for the mount point, otherwise just the default demo code.  Very likely I am just misunderstanding something basic.  The two log files seem to suggest that at least the initial steps in mounting the blank and named volumes are similar.  The first indication of something odd is when the named volume reports no space available.

Thanks again for your continued assistance!
-marc

USB HOST READY.
USB HOST state 01
(mounted) fs_ready() = 1, wasmounted = 0, PFAT_VOLUMES = 1
No media. Waiting to mount /foo
USB HOST state 02
USB HOST state 0a
USB HOST state 03
USB HOST state 04
USB HOST state 05
USB HOST state 0c
USB HOST state 0d
USB HOST state 60
(mounted) fs_ready() = 0, wasmounted = 1, PFAT_VOLUMES = 1
/foo mounted, fs_ready() = 0

USB HOST READY.
USB HOST state 01
(mounted) fs_ready() = 1, wasmounted = 0, PFAT_VOLUMES = 1
No media. Waiting to mount /
USB HOST state 02
USB HOST state 0a
USB HOST state 03
USB HOST state 04
USB HOST state 05
USB HOST state 0c
USB HOST state 0d
USB HOST state 60
(mounted) fs_ready() = 0, wasmounted = 1, PFAT_VOLUMES = 1
/ mounted, fs_ready() = 0
Removing '/HeLlO.tXt' file... completed with 4

Starting Write test...
File opened OK, fd = 1
Wrote 19 bytes, File closed result = 0.

Starting Read test...
File opened OK, fd = 1, displaying contents...
]-[ello \/\/orld!

Read completed, last read result = -1 (20), file close result = 0.
Testing rename
file rename result = 0.

Removing '/1MB.bin' file... completed with 0
1MB write timing test  2048 writes, (0), (0),  14377 ms (14 sec)
completed with 0
1MB read timing test 2048 reads, (20),  11786 ms (12 sec)
completed with 0
Directory of '/'
-rw--a      1048576 2015-11-14 23:39:28 1MB.bin
-rw--a           19 2015-11-14 23:39:14 newtest.txt
1072381952 bytes available on disk.

Flushing caches...
Remove and insert media...

//#define USB_VOL_NAME  "/"
#define USB_VOL_NAME  "/foo"
//#define USB_VOL_NAME  "/fo1"

uint8_t current_state = 128;
uint8_t last_state = 255;

void loop() {
#if !USB_HOST_SHIELD_USE_ISR
        MAX3421E_Usb.Task();
#endif

#if 1
        current_state = MAX3421E_Usb.getUsbTaskState();
        if(current_state != last_state) {
                last_state = current_state;
                printf("USB HOST state %2.2x\r\n", current_state);
        }
#endif
        if(mounted != wasmounted) {
               wasmounted = mounted;
                        uint64_t fre;
#if MAKE_BIG_DEMO
                        uint32_t start;
                        uint32_t end;
                        uint32_t wt;
                        uint32_t rt;
#endif
                        int res;
                        int fd;
#if MAKE_BIG_DEMO
                        fre = fs_getfree(USB_VOL_NAME);
                        if(fre > 2097152) {
                                printf_P(PSTR("Removing '/HeLlO.tXt' file... "));
                                fflush(stdout);
								
 
                         } else {
#endif

                } else {
                        printf("No media. Waiting to mount %s\r\n", USB_VOL_NAME);
                }
        }
}

xxxajk, 2015-10-15:
Will check it out soon.

xxxajk, 2015-10-15:
Will push a demo that will be more clear for you to use.
You are a touch confused on the relationship with volume names and path names. :-)

marcah, 2015-10-15:
I am glad to hear that you have an understanding of “where I have wandered off into the weeds.”
Thanks!

xxxajk, 2015-10-15:
Start.

USB HOST READY.
USB HOST state 1d
No media. Waiting to mount /foo
USB HOST state 02
USB HOST state 0a
USB HOST state 03
USB HOST state 04
USB HOST state 05
USB HOST state 0c
USB HOST state 0d
USB HOST state 60
/foo mounted.
Removing '/foo/HeLlO.tXt' file... completed with 4

Starting Write test...
File opened OK, fd = 1
Wrote 19 bytes, File closed result = 0.

Starting Read test...
File opened OK, fd = 1, displaying contents...
]-[ello \/\/orld!

Read completed, last read result = -1 (20), file close result = 0.
Testing rename
file rename result = 0.

Removing '/foo/1MB.bin' file... completed with 0
1MB write timing test  2048 writes, (0), (0),  10631 ms (11 sec)
completed with 0
1MB read timing test 2048 reads, (20),  10648 ms (11 sec)
completed with 0
Directory of '/foo/'
-rw--a      1048576 2015-11-15 15:20:50 1MB.bin
-rw--a           19 2015-11-15 15:20:40 newtest.txt
-rw--a 531044984 2014-07-15 02:00:06 ASTUDI~1.EXE (AStudio6_2sp1_1502.exe)
-rw--a        38936 2015-01-16 04:04:52 orn.ps
-rw--a        92576 2015-05-26 06:01:26 switch.jpg
-rw--a        66180 2015-05-26 06:01:38 regular.jpg
507559936 bytes available on disk.

Flushing caches...
Remove and insert media...

Pushing new demo now. B-) let me know if you have success with it.

marcah, 2015-10-15:
Hey Andrew,
Thanks so much!  I actually noticed that you had pushed the change before reading your message.  I have tried all three of the test drives; as you would expect, they all work as expected.

I’m not sure how long it would have taken me to figure out my misunderstanding.  I indeed appreciate your assistance.

I have also tried to connect the three drives via a 4 port hub.  The order of the drives seems to matter in terms of trying to mount a specific drive.  Seems that the other drives can be on one side of the target drive, perhaps “behind” in some ordering.  If located on the “leading” side, the target drive is not found.  Could it possibly be the case that I have not actually loaded the hub library as I don’t have an inline define as with the other libraries?

It seems that I may not be far from having all of the fundamental functionality in place to support my application.  When able to mount two drives simultaneously I will be very close.

Thanks again for your (and others) investments to “hide” so much of the complexity related to the use of USB.  UHS30 is indeed a very valuable contribution!

Regards,
-marc

marcah, 2015-10-16:
Hey Andrew,
Further success & observations.

I have increased PFAT_VOLUMES from the default of 1 to 4 and now am able to mount more than one drive.  There does still seem to be some sort of ordering that I don’t fully understand.  For instance, if I attempt to check for three devices and the second is not present, I don’t seem to be able to find the third.  With a list of three, if the first two are present, I seem to be able to mount both of them.  While I don’t fully understand what I am observing, I am very encouraged with regard to how close I am to demonstrating the functionality required.

Thanks again!

From: Marc Hartquist
Sent: Thursday, October 15, 2015 2:17 PM
To: 'felis/UHS30'
Subject: RE: [UHS30] Mount point for USB drive (#8)

Hey Andrew,
Thanks so much!  I actually noticed that you had pushed the change before reading your message.  I have tried all three of the test drives; as you would expect, they all work as expected.

I’m not sure how long it would have taken me to figure out my misunderstanding.  I indeed appreciate your assistance.

I have also tried to connect the three drives via a 4 port hub.  The order of the drives seems to matter in terms of trying to mount a specific drive.  Seems that the other drives can be on one side of the target drive, perhaps “behind” in some ordering.  If located on the “leading” side, the target drive is not found.  Could it possibly be the case that I have not actually loaded the hub library as I don’t have an inline define as with the other libraries?

It seems that I may not be far from having all of the fundamental functionality in place to support my application.  When able to mount two drives simultaneously I will be very close.

Thanks again for your (and others) investments to “hide” so much of the complexity related to the use of USB.  UHS30 is indeed a very valuable contribution!

Regards,
-marc

xxxajk, 2015-10-16:
Give each volume a unique name.
E.g.
d1
d2
d3
you should see only the volumes available.

marcah, 2015-10-16:
I am now able to mount two drives and open 4 logfiles per drive.  Next step is to write the actual data to be logged.

The very subtle difference between ‘vol_label’ and ‘vol_path’ that you clarified through the updated demo was key to my moving forward and will likely benefit others going forward.

Your assistance is of great value to me.  I would like to discuss a means that will allow me to directly support your (and team) continued development and support.  Should you choose, you may contact me directly at marc@firststrand.com<mailto:marc@firststrand.com> .

Best regards,
-marc

marcah, 2015-10-17:
Hello Andrew,
I continue to make progress in exploiting the capabilities of UHS30.  I am mounting and writing data to a non-blank labeled drive.  Again, thanks for the updated demo that made my error obvious.

I am currently having trouble appending to a file.  The following is a portion of UHS_FS_INLINE.h:

                rv = _fs_open(pathname, &mode);
                if((rv == 0) && (flags & O_CREAT)) {
                        // try to create
                        rv = _fs_open(pathname, &mode2);
                }
                if(rv != 0) {
                        if(flags & O_WRONLY) {
                                switch(flags & (O_APPEND | O_TRUNC)) {
                                        case O_APPEND:
// added MAH 10/16/2015
printf("O_APPEND mode\r\n");
                                                // seek to eof
                                                s = fs_lseek(rv, 0, SEEK_END);
                                                if(s) {
                                                        fs_close(rv);
                                                        rv = -1;
                                                }
                                                break;

I added the above statement to verify that I am indeed “landing in the right place.”  I am opening the file w/ “O_WRONLY | O_APPEND”.

You may recall that in addition to UHS30, I am using the SD library.  I have been dealing with some integration issues as UHS30 includes what appears to be some leveraged SD code.  This issue may turn out to be an issue with integration changes that I have made.  In any case, I can write to the USB drive but cannot seem to append.

Regards,
-marc

xxxajk, 2015-10-17:
No parts of sd lib were used, but they may have common ancestors...

Try a test that opens in mode O_RDWR | O_APPEND instead, because if you
really think about it, you can't seek on a write only file. Seek requires
read.

xxxajk, 2015-10-17:
BTW: the FAT FS implementation in the SD lib is pretty buggy, and is prone to FAT table corruption, whereas mine isn't ;-) You may be better off merging SD, by hooking into a custom FS library.
Look for comments about adding your own drivers within the source.

marcah, 2015-10-17:
Hey Andrew,
Thanks for the assistance!  I performed a brief test using O_RDWR | O_APPEND… thus far, same result.  I plan to perform a test using the base demo code outside of my application as a means to better understand the issue.  I suspect that I have an integration issue that may be leading to the append failure.

BTW… did you see my response at the end of one of our threads on 10/2 related to my observations regarding the inline logic?
Regards,
-marc

xxxajk, 2015-10-18:
Indeed it smells like a collision. If it works without any SD library code, then the answer is simple. You just need to extend the UHS one. It is easy top do, honest.

xxxajk, 2015-10-18:
Here are some hints, they're right in the code, and (at least to me) quite obvious.
Basically, just look for code that mentions __UHS_BULK_STORAGE_H__
You'll need to duplicate those pieces.
Here are some references for you to see for yourself, where you would do stuff.
https://github.com/felis/UHS30/blob/master/libraries/UHS_FS/UHS_FS.h#L140
https://github.com/felis/UHS30/blob/master/libraries/UHS_FS/UHS_FS_INLINE.h#L215
https://github.com/felis/UHS30/blob/master/libraries/UHS_FS/UHS_FS_INLINE.h#L233

Just provide the few methods needed in your class, then extend it.

One caveat I can think of though is that SD doesn't do anything in the background, so anything done, like checking insertion/removal, mounting may be a bit of a hassle, unless you can have SD generate IRQs, other than that, the rest of the code is pretty generic.
I'd suggest just using one of those tiny microSD readers, and go full USB. It would be so much easier.

marcah, 2015-10-18:
Thanks for the pointers.  I am out at a conference this week.  I am hoping to at least verify ‘append’ with the generic demo before I leave.  With the results of the test, I can then figure out if I need to integrate some basic SD functionality as you suggest.  I will not require much in the way of SD functionality.  The system will essentially never need to deal with insertion / removal as there will be no physical access to the SD card at all.  The USB drives will be periodically swapped.

Regards,
-marc

https://github.com/felis/UHS30/blob/master/libraries/UHS_FS/UHS_FS.h#L140
https://github.com/felis/UHS30/blob/master/libraries/UHS_FS/UHS_FS_INLINE.h#L215
https://github.com/felis/UHS30/blob/master/libraries/UHS_FS/UHS_FS_INLINE.h#L233

marcah, 2015-10-23:
Hello Andrew,

I have performed a few tests and have some results that I would like to share with you.

Setup:

1)      Removed all of the Atmel related tools.

2)      Uninstalled Arduino 1.6.5r2 and installed Arduino 1.6.5r5 (latest version).

3)      Refreshed UHS30 that includes changes from a week ago.

4)      Using UHS_FS_NEW_DEMO with minor adjustments for all tests.

5)      Two 8GB USB drives each with a 1 GB FAT partition

a.      /

b.      /thl02

6)      Mega2560 + screw shield + USB shield + RTC + 4 port USB Hub

Append Test:

1)      Using O_RDWR | O_APPEND works as expected to append to an existing file.

2)      No SD hardware or associated libraries loaded.

3)      Indicates that indeed my full application has a defect preventing append to function correctly.

Multiple USB Drives:

1)      Each USB 1GB partitioned drives works when connected to the hub w/o any other drives.

2)      Each USB drive works if located “before” the other drive connected to the hub.

a.      fs_ready(“/”)

b.      Hub port 1 -> ‘/’, ‘/’ mounts fine

c.      Hub port 2 -> ‘/thl02’

3)      Each USB drive fails if located “after” the other drive connected to the hub.

a.      fs_ready(“/”)

b.      Hub port 1 -> ‘/thl02’

c.      Hub port 2 -> ‘/’, ‘/’ times out “USB HOST state 60”

4)      I am likely misunderstanding something.  However, it appears that the order of the USB devices plugged into the hub matters in terms of mounting.

5)      Each system will have two USB storage devices connected.  There are a total number of 10 devices labeled “thl01” – “thl10”.   As the devices need to be uniquely labeled, I am pursuing a strategy of having a list of all 10 possible modules and checking ‘fs_ready(“thlxx”)’ for each device until I identify the two connected devices.

Summary:

1)      Thanks for the suggestion to try O_RDWR instead of O_WRONLY in the case of append.

2)      Any thoughts as to what I could be overlooking in terms of the observed “ordering dependency?”

Regards,
-marc

From: Marc Hartquist
Sent: Sunday, October 18, 2015 9:50 AM
To: 'felis/UHS30'; felis/UHS30
Subject: RE: [UHS30] Mount point for USB drive (#8)

Thanks for the pointers.  I am out at a conference this week.  I am hoping to at least verify ‘append’ with the generic demo before I leave.  With the results of the test, I can then figure out if I need to integrate some basic SD functionality as you suggest.  I will not require much in the way of SD functionality.  The system will essentially never need to deal with insertion / removal as there will be no physical access to the SD card at all.  The USB drives will be periodically swapped.

Regards,
-marc

https://github.com/felis/UHS30/blob/master/libraries/UHS_FS/UHS_FS.h#L140
https://github.com/felis/UHS30/blob/master/libraries/UHS_FS/UHS_FS_INLINE.h#L215
https://github.com/felis/UHS30/blob/master/libraries/UHS_FS/UHS_FS_INLINE.h#L233

xxxajk, 2015-10-23:
See your private email. I'll then present an answer here.

marcah, 2015-10-29:
Hello Andrew,

I have indeed verified that the O_RDWR | O_APPEND is working as expected.  The issue was as I suspected on my end.  Turns out that many of the mode flags used by UHS30 and SD are very similar, but upon closer inspection, not identical.  In combining both UHS30 and SD, there were a few symbols that were redefined.  I had undefined a few of the duplicates after they had been referenced.  Once redefined, they were fine for SD and wrong for UHS30.  For now, I have adjusted the SD definitions to match UHS30 and all seems to be functioning as expected.

Do you anticipate having a chance to look at the USB device ordering issue?

Best regards,
-marc

xxxajk, 2015-10-29:
Been quite busy, will have time later today.

marcah, 2015-11-04:
Hello Andrew,
Have you had a chance to look at the “USB device ordering” issue?  If not, I can probably more thoroughly characterize the issue.  My project is approaching a point where I will need to identify a path through or around the issue.  It is very likely that I am simply misusing UHS30 functionality.

Best Regards,
-marc

xxxajk, 2017-01-18:
Fixed with email discussions.

closed 2017-01-18 by xxxajk