felis/UHS30 #9: Due to Due Serial Communication Via USB Possible?

Delphi, 2015-11-04:
Greetings to the authors of this library. 

First of all, thank you for taking the time and efforts to create a promising library!

What I am trying to achieve:-
I am currently working on a project that uses nearly every single pin header available on a Due, including all hardware serial ports and I would like to use the native USB port on the Due's to communicate between each other.

I understand that it is possible to setup one Due as a USB client serial device by invoking the SerialUSB.begin(9600); command and then sending data using SerialUSB.println("Hello World");
When connected to my PC or connected to my android phone using a OTG cable with the phone as a host, it works a treat.

As such I would require to setup the master Due as a USB host with an OTG cable to accept the CDC? connection/device that is the slave Due and to be able to read the serial data using SerialUSB.read();

It would appear that all the serial communication commands are there already, however I am struggling to get the Host Due to open a serial port with the slave Due.

My Question:-
Is this library, in its current form, possible to achieve what I am wishing to?
I have downloaded it and have played with it some, however I am not familiar with setting up the native USB host on the Due so my efforts have been to no avail thus far.

If the library does not currently support this type of connectivity, is this something that is planned for this library in the future? 

Again many thanks for your time and effort with this library as I foresee it opening the doors to all sorts of cool project ideas!

I understand this project is not yet in a position to start providing technical support, as mentioned on the home page, however if you can provide any pointers that would get me a step closer to my goals it would be hugely appreciated.

Many thanks.
Delphi.

xxxajk, 2015-11-04:
Have to write CDC-ACM drivers for this yet, and due native is not
implemented yet.

Delphi, 2015-11-04:
Thank you for the fast response!
I shall keep an eye on this project to monitor for updates.

Thank you.

wesner0019, 2016-03-28:
Hi xxxajk, 

I am also very interested in this type of function. I am wondering if there is any progess on this for the due native support? Is the any type of donations I could make to help? I have been trying many things with the 2.0 library but I have limited programming experience.

Thanks,
Steve

xxxajk, 2016-03-28:
Sorting out MIPS32 (PIC32MZXXX) right now, but it is in the to-do queue. If
you want to try your luck at it, help is welcomed.

wesner0019, 2016-03-28:
Thanks, I am actually wanting to use this library and the shield so that I can use the above functionality to communicate between two STM32F205 120Mhz ARM Cortex M3 (Particle P1 modules, https://docs.particle.io/datasheets/p1-datasheet/). Would this type of processor require its own native support?

xxxajk, 2016-03-28:
Short answer: yes.

wesner0019, 2016-03-28:
Ok thanks, I'm in a small start-up in Minnesota and my team is really looking at this type of functionality as a critical feature but we are not experts with USB communication protocol. We would be willing to pay you/your team and send you a few Photons to get this type of functionality going rather soon. 

If you're open to coding this please PM me at: wesner0019@gmail.com 

Thanks,
Steve

felis, 2016-03-28:
Hi Steve,

I sent you an e-mail.

Oleg.

kubitron, 2016-09-06:
Hello all.  Sorry if this is a dead thread, but I'm curious if support for Arduino Zero (same as Due?) has happened yet.  Very interested in Host CDC serial.

Inquiring minds always want to know.  I'm also very happy to work with early code.

kubitron, 2016-09-06:
(p.s. I asked, above, since I see that CDC-ACM support seems to exist now..., so just wondering about native support for Arduino Zero).

xxxajk, 2016-09-06:
Time is limited at present. Nothing stops you from contributing.

kubitron, 2016-09-11:
Indeed.  Just curious what was up with Zero support.  I had to come up with another solution for current project, but perhaps could help at point in future.  Any suggestions on how to become sufficiently knowledgeable about USB/your code to be helpful?

avanshvishu99, 2020-06-16:
I also want make my android as arduino server and also want to use its sensors to function the arduino with only otg usb support without any extra shield

avanshvishu99, 2020-06-24:
Can we use blynk usb script in android old otg...to give arduino internet...and use all its sensors.,..and control with my new phone

xxxajk, 2020-06-24:
If you want, you can port in support for due. Should not be difficult. I
just lack the time to do it. Pull requests welcome.