felis/UHS30 #10: PR 185 from 2.0

Lauszus, 2015-11-18:
We need to make the same fix as in PR: https://github.com/felis/USB_Host_Shield_2.0/pull/185/files.

A quick search shows the relevant files: https://github.com/felis/UHS30/search?utf8=%E2%9C%93&q=epAttribs. I recommend never setting epAttribs explicit to zero, as this will cause bugs if bmNakPower is already set.

felis, 2015-11-18:
hold on on this till the next week. I'm building a permanent test setup so that any fix can be checked quickly against the real gear. It will be a linux machine with several arduino/shield combos always connected. The hardware for the box will arrive Friday, I'll put it online over the weekend.

xxxajk, 2018-02-16:
Been a long time for me to actually look into this, and....

Does not apply!

bmNakPower is never cleared or set by the driver code. 

Attributes are only reset in the proper places within the core enumeration code and bmNakPower is always set correctly after the clearing of the toggles.

Setting the attributes to zero is faster and less code than setting each individual member of the union.
1 instruction v.s. 2!

So... closing it, 'tis a non-issue.

closed 2018-02-16 by xxxajk

Lauszus, 2018-02-16:
It was an issue for the ACM code: https://github.com/felis/USB_Host_Shield_2.0/issues/184. I just decided to change all of them, so people didn't accidentally clear ```bmNakPower``` when copying pasting code around.

xxxajk, 2018-02-16:
That's the advantage of having such a thing in one place as on UHS3 ;-) Perhaps UHS2 could use a common inline call instead of having it sprinkled all over the place, so that the order is always correct. That way, any copy/paste won't be an issue.