felis/UHS30 #13: Headers only include of submodules

markmaker, 2016-04-13:
Hi

now that the lib was restructured with submodules, shouldn't there be a different define for headers only include?

If I #define i.e. `LOAD_UHS_KINETIS_FS_HOST` both `UHS_KINETIS_FS_HOST.h` and `UHS_KINETIS_FS_HOST_INLINE.h` will be included. If I don't define anything, nothing will be included.

So it's not possible to use UHS3 from different source files, i.e. I'd like to put all the USB host stuff in a different source file away from the .ino

Thanks,
Markk

xxxajk, 2016-04-13:
No. The defines go in the sketch. 
That is the entire idea -- not compiling anything you don't need and do not ask for.
That said, you should be able to use a header file in your sketch folder containing the defines and the include just to load them.
If you require getting to any of the information within the headers, without including the compiled code as well, simply do the defines needed and include the main header. This will get you what you want.

markmaker, 2016-04-14:
> If you require getting to any of the information within the headers, without including the compiled code as well, simply do the defines needed and include the main header. This will get you what you want.

But how? What defines? [I only see the LOAD defines in UHS_host.h](https://github.com/felis/UHS30/blob/master/libraries/UHS_host/UHS_host.h#L63). It seems all-or-nothing.

```
// Load USB drivers and multiplexers

#if defined(LOAD_UHS_HUB)
#include "UHS_HUB/UHS_HUB.h"
#endif // HUB loaded

#if defined(LOAD_UHS_BULK_STORAGE)
#include "UHS_BULK_STORAGE/UHS_BULK_STORAGE.h"
#endif

#if defined(LOAD_GENERIC_STORAGE)
#include "../UHS_FS/UHS_FS.h"
#endif

```
Thanks 

-Markk

xxxajk, 2016-04-14:
What I meant is to do is like this:

file a_test.ino:
```

#include <Arduino.h>
#ifdef true
#undef true
#endif
#ifdef false
#undef false
#endif

#include "UHS_stuff.h"

void setup() {
        while(!USB_HOST_SERIAL);
        USB_HOST_SERIAL.begin(115200);

        E_Notify(PSTR("\r\n\r\nStarting CDC-ACM test program...\r\n"), 0);
        while(MAX3421E_Usb.Init(1000) != 0);
        E_Notify(PSTR("\r\n\r\ngo!\r\n"), 0);

}

void loop() {

        if(Acm.isReady()) {
                uint8_t rcode;

                /* reading the keyboard */
                if(USB_HOST_SERIAL.available()) {
                        uint8_t data = USB_HOST_SERIAL.read();
                        /* sending to the phone */
                        rcode = Acm.Write(1, &data);
                        if(rcode)
                                ErrorMessage<uint8_t>(PSTR("SndData"), rcode);
                }//if(Serial.available()...

                /* reading the phone */
                /* buffer size must be greater or equal to max.packet size */
                /* it it set to 64 (largest possible max.packet size) here, can be tuned down
                for particular endpoint */
                uint8_t buf[64];
                uint16_t rcvd = 64;
                rcode = Acm.Read(&rcvd, buf);
                if(rcode && rcode != hrNAK)
                        ErrorMessage<uint8_t>(PSTR("Ret"), rcode);

                if(rcvd) { //more than zero bytes received
                        for(uint16_t i = 0; i < rcvd; i++) {
                                USB_HOST_SERIAL.print((char)buf[i]); //printing on the screen
                        }
                }
        }
}

```

... and you place the UHS stuff in a header, like this...

file UHS_stuff.h:
```
#define LOAD_UHS_PRINTF_HELPER
#define LOAD_USB_HOST_SYSTEM
#define LOAD_USB_HOST_SHIELD
#define LOAD_UHS_CDC_ACM
#define LOAD_UHS_CDC_ACM_XR21B1411
#define LOAD_UHS_CDC_ACM_PROLIFIC
#define LOAD_UHS_HUB

#include <UHS_host.h>

MAX3421E_HOST MAX3421E_Usb;
UHS_USBHub hub_MAX3421E(&MAX3421E_Usb);

class MY_ACM : public UHS_CDC_ACM {
public:

        MY_ACM(UHS_USB_HOST_BASE *p) : UHS_CDC_ACM(p) {
        };
        uint8_t OnStart(void);
};

uint8_t MY_ACM::OnStart(void) {
        uint8_t rcode;
        // Set DTR = 1 RTS=1
        rcode = SetControlLineState(3);

        if(rcode) {
                ErrorMessage<uint8_t>(PSTR("SetControlLineState"), rcode);
                return rcode;
        }

        UHS_CDC_LINE_CODING lc;
        lc.dwDTERate = 9600;
        lc.bCharFormat = 0;
        lc.bParityType = 0;
        lc.bDataBits = 8;

        rcode = SetLineCoding(&lc);

        if(rcode)
                ErrorMessage<uint8_t>(PSTR("SetLineCoding"), rcode);

        return rcode;
}

MY_ACM Acm(&MAX3421E_Usb);

```

Doing other stuff is VERY advanced programming.

markmaker, 2016-04-14:
I think I found a solution. In a file where I only wanted the declaration but not the implementation I could:

```
...
#define LOAD_UHS_BULK_STORAGE
#define UHS_BULK_STORAGE_LOADED // implementation already loaded elsewhere
...
#include <UHS_host.h> // UHS USB HOST base classes

```

> Doing other stuff is VERY advanced programming.

I agree that including everything into the .ino allows one to work around the strange fact, that Arduino (to my knowledge) has no concept of setting overall #defines per a sketch (i.e. by having a `configure.h` in the sketch folder that is force-included into every compilation including all libs). 

However this `INLINE` style is still a hack that goes against many advantages of having separate .cpp files. 

- modularization of your project
- encapsulation (hiding stuff you don't want the users of your module or lib to see or use by accident)
- separating name spaces both for the preprocessor and C/C++ `static` (i.e. [internal linkage](https://en.wikipedia.org/wiki/Linkage_%28software%29)) symbols preventing unnecessary conflicts across modules or libs 
- allowing for module _implementation_ inter-dependencies that are cyclic (non-[DAG](https://en.wikipedia.org/wiki/Directed_acyclic_graph)) which at least in project modules can hardly be prevented
- much faster compilation i.e. compile only modules or libs that have changed 

At least with Visual Micro this last bit seems to matter much - ever since I started with UHS3, compilation is painfully slow. 

But again: I see the rationale in the Arduino playground.

_Markk

xxxajk, 2016-04-15:
Yes, that is the advanced way to do this.

closed 2018-02-16 by xxxajk