felis/UHS30 #14: Does it work with ADK?

rwhejny, 2016-08-13:
Hi,
I would like to try this out!  I am currently using the USB Host 2.0 library with Android ADK and am finding issues with the Usb.Task().  I think an interrupt based usb host would work better.

Is there a UHS3.0 example that would work with ArduinoBlinkLED for Android, like there is for the UHS2.0?  If not, any other examples that work with ADK?

Thanks!

xxxajk, 2016-08-13:
Should work, try it.

rwhejny, 2016-08-14:
Ok.

Quick first question.  What is the replacement of adk.RcvData and adk.SndData in the UHS3 library?

xxxajk, 2016-08-14:
If you are expecting the accessory mode to work, the interface driver for it is not yet written. 
You are welcome to help out, though.
What should work:  https://github.com/felis/UHS30#current-working-interface-drivers

xxxajk, 2016-08-14:
BTW, Don't cut yourself short on skills. 
**# ANY** kind of help is appreciated.

xxxajk, 2017-01-17:
Now In progress with help.

xxxajk, 2017-01-18:
ADK is now working.
Please let me know if it works for you.

shadeyinka, 2017-09-08:
UHS20 works with Arduino Mega ADK but UHS30 seem not to work.UHS_FS/UHS_HOST_SHIELD/UHS_FS_NEW_DEMO
Compiles OK But
Execution gets stuck at while(KINETIS_Usb.Init(1000) != 0);

xxxajk, 2018-02-16:
Arduino Mega ADK now supported...

closed 2018-02-16 by xxxajk