felis/UHS30 #23: UHS_FS_NEW_DEMO on USB Host Shield and Teensy 3.5

mjs513, 2017-01-23:
This is a continuation of the issue raised on ush2 with accessing a mass storage device using a usb host shield and a teensy 3.5.  I incorporated the suggestions in the last post:

```
#define UHS_DEVICE_WINDOWS_USB_SPEC_VIOLATION_DESCRIPTOR_DEVICE 0

// Uncomment to debug
#define ENABLE_UHS_DEBUGGING 1
// Uncomment to make debugging very noisy
#define DEBUG_PRINTF_EXTRA_HUGE 1
```
and still hangs.  I am attaching a text file with the debug messages for your reference.

[Teensy35 Debug Dump.txt](https://github.com/felis/UHS30/files/724892/Teensy35.Debug.Dump.txt)

Mike

xxxajk, 2017-01-23:
try
`
#define DEBUG_PRINTF_EXTRA_HUGE_UHS_HOST 1
`
and show me the output from that (warning there will be quite a lot of data)

mjs513, 2017-01-23:
Ok will run it now.  What FS LOCK do you want me to use?

xxxajk, 2017-01-23:
For 1 file use 1 lock, for 2, use 2, etc.
The lock just determines how many files can be opened at the same time.

mjs513, 2017-01-23:
Yep got that, but the sketches performance with thumb drive (how many errors I get) seems to be dependent on that value.  Running it now for 1024

xxxajk, 2017-01-23:
Yes, that value also tweaks how many buffers are used.
Be sure to use `fs_sync();` before ejecting media in your own sketches.

mjs513, 2017-01-23:
Got it thanks.  Now just to get through this issue.

closed 2017-05-27 by mjs513