felis/UHS30 #24: Logitech Extreme 3D Pro Joystick Compatibility

simonnarang, 2017-02-26:
I am making a robotic arm. The controller I am using is a Logitech extreme 3D pro. I was using an Arduino Mega 2560 with the Arduino USB Host Shield to connect to the joystick. In order to get more processing power, I switched to the Arduino Due, which has a native USB port built in. This native port replaces the need for the Arduino USB Host Shield. Because the [USB_Host_Shield_2.0 Library](https://github.com/felis/USB_Host_Shield_2.0) is built for the Arduino USB Host Shield ***ONLY***. Because of this, I need something (like this library) than can connect the Logitech Extreme 3D Pro Joystick to the Arduino Due. Is there an example [like the one in USB_Host_Shield_2.0](https://github.com/felis/USB_Host_Shield_2.0/tree/master/examples/HID/le3dp)?

| Old Setup with Arduino Mega & USB Host Shield | New Setup with Arduino Due |
|--|--|
| [image] | [image] |

xxxajk, 2017-03-01:
Not yet...
The HID API has not been worked out, and many things need to be considered on how to integrate HID.
HID gets complicated when Bluetooth is added to the mix as well.

rodan, 2019-02-17:
check out the LUFA project: http://www.fourwalledcubicle.com
it's got two examples of USB HOSTs used to read HID joysticks.
their HID report parser is great too.

xxxajk, 2019-02-17:
No need. We are developing a sane API here, eventually, just called HID-JOYSTICK, which will simply encompass all the different joysticks in a way that they all can be accessed with the same user code no matter which one is plugged in.