felis/UHS30 #28: UHS MIDI Driver issue

mattomatto, 2017-04-04:
Hi there, 

I know this is still in development, so forgive me if you are already aware of this.

Testing various MIDI controllers with this library, and I can only get one to work.  Looking at the debug for those that don't work (below), it appears that the class doesn't find a compatible library, however these controllers were definitely supported by UHS v2, and are class compliant.

Thanks.

Testing Arturia Keystep:
```
Configuring: parent = 0, port = 1, speed = 1

Configuring PktSize x40,  rcode: x00, retries 0,
ctrlReq2: left: 64, read:64, nbytes 64
ctrlReq3: acceptBuffer sz 18 nbytes 64 left 46
Address: 1 (0.0.1)
0 retries.
DevDescr 2nd poll, bMaxPacketSize0:64
ctrlReq2: left: 18, read:18, nbytes 18
Address: 1 (0.0.1)
0 retries.
configs: 1
ctrlReq2: left: 9, read:9, nbytes 9
CONFIGURATION: 0, bNumInterfaces 2, wTotalLength 101
bLength: 9 bDescriptorType: 02
eating 8
bLength: 9 bDescriptorType: 04
INTERFACE DESCRIPTOR FOUND
Getting 0 endpoints
eating 1
ENDPOINT DESCRIPTORS FILLED
TestInterface VID:1c75 PID:0288 Class:00 Subclass:00 Protocol 00
Interface data: Class:01 Subclass:01 Protocol 01, number of endpoints 0
Parent: 00, bAddress: 01
No driver at index 1
No driver at index 2
No driver at index 3
No driver at index 4
No driver at index 5
No driver at index 6
No driver at index 7
No driver at index 8
No driver at index 9
No driver at index 10
No driver at index 11
No driver at index 12
No driver at index 13
No driver at index 14
No driver at index 15
No driver for this interface.
bLength: 36 bDescriptorType: 01
eating 35
bLength: 9 bDescriptorType: 24
eating 8
bLength: 9 bDescriptorType: 24
eating 8
bLength: 9 bDescriptorType: 05
eating 8
bLength: 5 bDescriptorType: 25
eating 4
bLength: 9 bDescriptorType: 05
eating 8
bLength: 5 bDescriptorType: 25
eating 4
```

Testing AKAI LPK25: 

```
Configuring: parent = 0, port = 1, speed = 1

Configuring PktSize x40,  rcode: x00, retries 0,
ctrlReq2: left: 64, read:64, nbytes 64
ctrlReq3: acceptBuffer sz 18 nbytes 64 left 46
Address: 1 (0.0.1)
0 retries.
DevDescr 2nd poll, bMaxPacketSize0:32
ctrlReq2: left: 18, read:18, nbytes 18
Address: 1 (0.0.1)
0 retries.
configs: 1
ctrlReq2: left: 9, read:9, nbytes 9
CONFIGURATION: 0, bNumInterfaces 3, wTotalLength 133
bLength: 9 bDescriptorType: 02
eating 8
bLength: 9 bDescriptorType: 04
INTERFACE DESCRIPTOR FOUND
Getting 2 endpoints
bLength: 9 bDescriptorType: 21
eating 7
bLength: 7 bDescriptorType: 05
ENDPOINT DESCRIPTOR: 0
ENDPOINT DESCRIPTOR OK
eating 0
bLength: 7 bDescriptorType: 05
ENDPOINT DESCRIPTOR: 1
ENDPOINT DESCRIPTOR OK
eating 0
eating 1
ENDPOINT DESCRIPTORS FILLED
TestInterface VID:09e8 PID:003e Class:00 Subclass:00 Protocol 00
Interface data: Class:03 Subclass:00 Protocol 00, number of endpoints 2
Parent: 00, bAddress: 01
No driver at index 1
No driver at index 2
No driver at index 3
No driver at index 4
No driver at index 5
No driver at index 6
No driver at index 7
No driver at index 8
No driver at index 9
No driver at index 10
No driver at index 11
No driver at index 12
No driver at index 13
No driver at index 14
No driver at index 15
No driver for this interface.
bLength: 9 bDescriptorType: 04
INTERFACE DESCRIPTOR FOUND
Getting 0 endpoints
eating 1
ENDPOINT DESCRIPTORS FILLED
TestInterface VID:09e8 PID:003e Class:00 Subclass:00 Protocol 00
Interface data: Class:01 Subclass:01 Protocol 01, number of endpoints 0
Parent: 00, bAddress: 01
No driver at index 1
No driver at index 2
No driver at index 3
No driver at index 4
No driver at index 5
No driver at index 6
No driver at index 7
No driver at index 8
No driver at index 9
No driver at index 10
No driver at index 11
No driver at index 12
No driver at index 13
No driver at index 14
No driver at index 15
No driver for this interface.
bLength: 36 bDescriptorType: 01
eating 35
bLength: 9 bDescriptorType: 24
eating 8
bLength: 9 bDescriptorType: 24
eating 8
bLength: 9 bDescriptorType: 05
eating 8
bLength: 5 bDescriptorType: 25
eating 4
bLength: 9 bDescriptorType: 05
eating 8
bLength: 5 bDescriptorType: 25
eating 4
USB_INTERFACE END OF STREAM
```

YuuichiAkagawa, 2017-04-04:
Please check #25  and #27 .

mattomatto, 2017-04-04:
That did the trick.  Thanks!

closed 2017-04-04 by mattomatto