felis/UHS30 #30: PS4 Support

mjs513, 2017-05-27:
How do I get this to work with the ps4 controller?  Unstand in 2.0 but not in 3.0?

Thanks
Mike

mjs513, 2019-01-14:
Wondering if there is any timeline to getting this incorporated to the UHS3 library since it now supports T4

xxxajk, 2019-01-14:
Yes. Everything supported under UHS2 will be supported, and more.

xxxajk, 2019-01-14:
By the way, if you aren't aware, t3.6 ehci is getting close to perfection as well...
Thread here: https://forum.pjrc.com/threads/37615-Teensy-3-6-USB-Host-support

PaulStoffregen, 2019-01-24:
From the github activity, it's pretty clear version 2.0 is still the mainstream and still attracting virtually all the user base and most user contributions.  Maybe you're happy with that?  Maybe this is a conscious decision, not to promote 3.0 yet?

When you do want 3.0 to attract a user base, a first essential step getting USH30 included in Arduino's library manager, so users can actually install it.  The current structure (unlike all other Arduino libraries) and statements like "Pre-release of USB Host Library version 3.0. No technical support offered at this time.  This set of libraries is EXTREMELY ALPHA!" clearly communicate to users that they should stay with version 2.0.