felis/UHS30 #31: Question: Connect generic HID device directly to DUEs Native USB port?

HellGL, 2017-08-07:
Hi,
I have a USB HID device that I would like to connect directly to the Native USB port of my DUE. I know what data to expect and what data to send, but I don't know how to approach this problem. Is it at all possible to use this library for that purpose without the USB shield? Do I need to implement my own driver and if so, is there any example for doing this? Thanks for your help.

HellG.

---
Under linux my device reports the following via lsusb:
`Bus 001 Device 044: ID 0c29:020a  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x0c29 
  idProduct          0x020a 
  bcdDevice            1.00
  iManufacturer           0 
  iProduct                0 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0                                                                                                                                                     
    bmAttributes         0x40                                                                                                                                                     
      (Missing must-be-set bit!)                                                                                                                                                  
      Self Powered                                                                                                                                                                
    MaxPower                4mA                                                                                                                                                   
    Interface Descriptor:                                                                                                                                                         
      bLength                 9                                                                                                                                                   
      bDescriptorType         4                                                                                                                                                   
      bInterfaceNumber        0                                                                                                                                                   
      bAlternateSetting       0                                                                                                                                                   
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            8 French
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      63
          Report Descriptor: (length is 63)
            Item(Global): Usage Page, data= [ 0x00 ] 0
                            Undefined
            Item(Local ): Usage, data= [ 0x00 ] 0
                            (null)
            Item(Main  ): Collection, data= [ 0x01 ] 1
                            Application
            Item(Global): Report ID, data= [ 0x01 ] 1
            Item(Local ): Usage Minimum, data= [ 0x01 ] 1
                            (null)
            Item(Local ): Usage Maximum, data= [ 0x68 ] 104
                            (null)
            Item(Global): Logical Minimum, data= [ 0x00 ] 0
            Item(Global): Logical Maximum, data= [ 0x01 ] 1
            Item(Global): Report Count, data= [ 0x68 ] 104
            Item(Global): Report Size, data= [ 0x01 ] 1
            Item(Main  ): Input, data= [ 0x02 ] 2
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Global): Usage Page, data= [ 0x00 ] 0
                            Undefined
            Item(Local ): Usage, data= [ 0x00 ] 0
                            (null)
            Item(Global): Report ID, data= [ 0x01 ] 1
            Item(Local ): Usage Minimum, data= [ 0x01 ] 1
                            (null)
            Item(Local ): Usage Maximum, data= [ 0x38 ] 56
                            (null)
            Item(Global): Report Count, data= [ 0x38 ] 56
            Item(Global): Report Size, data= [ 0x01 ] 1
            Item(Main  ): Output, data= [ 0x02 ] 2
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Global): Report ID, data= [ 0x02 ] 2
            Item(Local ): Usage Minimum, data= [ 0x01 ] 1
                            (null)
            Item(Local ): Usage Maximum, data= [ 0x38 ] 56
                            (null)
            Item(Global): Report Count, data= [ 0x38 ] 56
            Item(Global): Report Size, data= [ 0x01 ] 1
            Item(Main  ): Output, data= [ 0x02 ] 2
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Global): Report ID, data= [ 0x03 ] 3
            Item(Local ): Usage Minimum, data= [ 0x01 ] 1
                            (null)
            Item(Local ): Usage Maximum, data= [ 0x18 ] 24
                            (null)
            Item(Global): Report Count, data= [ 0x18 ] 24
            Item(Global): Report Size, data= [ 0x01 ] 1
            Item(Main  ): Output, data= [ 0x02 ] 2
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Main  ): End Collection, data=none
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x000e  1x 14 bytes
        bInterval               4
Device Status:     0x0001
  Self Powered
`

xxxajk, 2017-08-07:
Short answer is yes.

Long answer, is that you need to fill in the hardware details to get it
working. UHS30 already supports more than just the classic USB Host
Shield.I would have done this at some point, but have been awfully busy.
You may be able to gleen a few hints from the kinetis_fs driver as far as
what is required.

HellGL, 2017-08-07:
Ok, lets start with: https://www.google.de/search?q=kinetis_fs ...
So short answer is rather: yes, but not for me.
But why is it not possible to start with the UHS_HID library? Sorry for the misaligned printout above. The item itself reports to be a HID device (bInterfaceClass 3 Human Interface Device) but I don't understand the HID example. I know my device sends 14 bytes per package of data to the host and can receive 20 bytes of configuration data, but I don't know where to send these...

xxxajk, 2017-08-07:
Because you need to actually be able to send packets out the native
interface before you can detect/enumerate/use/disconnect.

Unless you actually have another device currently supported, there's not
much sense in working on something you can't test/debug in quick cycles. I
suggest that to start off, that you either (for now) acquire either a uhs
board (works fine on the due) or grab a teensy3.2 in order to get used-to
the library.

For example, I still may need to fix the midi code submitted, probably
because it's not quite fitting in the abstract model that USB really is,
not because it isn't functional. It all has to do with having time, and
considering that I sometimes work 24/48/36 hour days, time can be a little
bit scarce currently. What little time I get, usually is for sleeping. I'm
hopeful that at some point, that normality will come forth and then I shall
be able to do more.

Briefly, if you do write something, it MUST be abstracted where it can just
be accessed from a common USB-HID function. That way, once bluetooth gets
moved in, a bluetooth keyboard/mouse will use the same exact entry point.
The idea is that you have a sketch, and don't care if the
keyboard/mouse/joystick/etc is on USB directly, or pushed over a bluetooth
connection via a bluetooth USB adaptor. The sketch would be able to use
either of them, without any actual code changes to the sketch, with
exception, perhaps, of an additional define which would select one/both/all
possible connections to a keyboard.

xxxajk, 2017-08-07:
p.s. the kinetis code i am referring to is here:
https://github.com/felis/UHS30/tree/master/libraries/UHS_host/UHS_KINETIS_FS_HOST

HellGL, 2017-08-08:
Thanks for your input. I found a XBOX Controller driver for USBHost library that I understand much better and I will try to derive my code from this example: http://forum.arduino.cc/index.php?topic=283058
 If this dosn't work as expected I will come back.

xxxajk, 2017-08-09:
It won't without a lot of porting, unless you actually have a board with the MAX3421E on it.