felis/UHS30 #32: Help to move data between SD Card & USB flash Drive with UHS30 or UHS20

shadeyinka, 2017-08-16:
I am building a datalogger whose primary memory is the SD card and the secondary storage is the USB flash drive. The data in the SD card is downloaded into the USB flash drive like once a week. I am using Arduino Mega ADK with Max3421 USB host chip.

1. How can I use both SD card and Flash Drive together. The excellent UHS30 library seems to conflict with SD.h, SdFat.h and Fatfs.h(to a limited extent).

Any example code will do, especially to be able to initialize and mount the SD card and write/read text files. CS on my SD card shield is on digital pin 8 (cs_pin=8) of my Arduino board.
2. I have some success with the SD card datalogger separately and UHS20 separately. UHS30 compiles on Arduino 1.6.9, AVR board 1.6.11, Windows 7 but it wouldn't execute any function (blank). Only UHS20 works for now.

xxxajk, 2017-08-16:
first important part from the example:
https://github.com/felis/UHS30/blob/master/libraries/UHS_FS/examples/USB_HOST_SHIELD/UHS_FS_NEW_DEMO/UHS_FS_NEW_DEMO.ino#L1

In other words you need to know the label on the target volume. No label defaults to "/", AKA root part of the filesystem tree.

UHS30's filesystem automatically will mount upon initial detection, and there is a way to know when that event happens. See here: https://github.com/felis/UHS30/blob/master/libraries/UHS_FS/examples/USB_HOST_SHIELD/UHS_FS_NEW_DEMO/UHS_FS_NEW_DEMO.ino#L273

After that point, you are free to read/write as needed. Just do not forget to close all files, and sync the filesystem when all writes are done. You do not need to unmount it, that happens automatically as well once you remove or power down the target storage.

shadeyinka, 2017-08-16:
Thanks a lot.
My problem wit UHS30 had been that code execution does not go beyond 
while(KINETIS_Usb.Init(1000) != 0);
the Aduino ADK just sits hear endlessly. Is there something i am doing wrong.
I use Arduino 1.6.8, Arduino AVR Board manager 1.6.11, Mega ADK Arduino, Windows 7.
Thanks a lot

xxxajk, 2017-08-16:
Sounds to me like a pin conflict, or I don't have something right for supporting the particular board.
I don't own one of these boards, so I can only make a guess that it is indeed a conflict.
If the SDcard library uses SPI, this indeed could be the case, especially if it does not use transactions.
Can you tell me which SDcard library you are using?

shadeyinka, 2017-08-16:
For now, the problem is with the UHS30 example files. Even without the SD card shield, the code hangs when executing the line

while(KINETIS_Usb.Init(1000) != 0);

UHS20 example testusbHostFAT.into works on my board. I don't know why UHS30 examples will not work.

Could it be the cs_pin for the max3421 chip? On my board, max3421 seems to be on pin 10, the SD card shield is on pin 8

xxxajk, 2017-08-16:
Ok, that is fine, what are the MISO/MOSI/CLK line pin numbers?

shadeyinka, 2017-08-16:
It uses the standard ICSP pins on Arduino mega

xxxajk, 2017-08-16:
Ok, did you connect 9 to pin 3 with a jumper wire?
https://github.com/felis/UHS30#avr-arduino

shadeyinka, 2017-08-16:
I did. Nothing happened

xxxajk, 2017-08-16:
Do you know where the INT pin from the MAX3421E is routed? 
Connect that to pin 3, and it should wake up and work. 
UHS30 uses interrupts and does all the work in the background, which means no polling...

shadeyinka, 2017-08-16:
I don't know where the INT pin of max3421 is, however, I will find out and do what you suggested.

I also noticed that the UHS20 with the examples from your github page didn't work either. I found one (probably modified) that ended up working. I will upload it to you in a few minutes. The difference might show something

xxxajk, 2017-08-16:
Try the following:
Connect pin 54 to pin 3
then... specify the SS/INT pins like so...
`MAX3421E_HOST MAX3421E(53, 3); // SS on pin 53, IRQ on pin 3
`

xxxajk, 2017-08-16:
...if that works, I'll place this fix in the library

shadeyinka, 2017-08-16:
I will check it out and give you a feedback if it works. Meanwhile, the link to the UHS2 that worked on my board is presented here:

https://github.com/Microduino/Microduino_Tutorials/tree/master/Microduino_Libraries/_06_Microduino_USBHOST_MAX3421E

Lauszus, 2017-08-16:
@xxxajk please see how I have done it in UHS2.0: https://github.com/felis/USB_Host_Shield_2.0/blob/ff8bdb24c19f254aa0496d3933920f3ae92f85a0/avrpins.h#L31-L32, https://github.com/felis/USB_Host_Shield_2.0/blob/ff8bdb24c19f254aa0496d3933920f3ae92f85a0/avrpins.h#L514-L517, https://github.com/felis/USB_Host_Shield_2.0/blob/ff8bdb24c19f254aa0496d3933920f3ae92f85a0/UsbCore.h#L37, and https://github.com/felis/USB_Host_Shield_2.0/blob/ff8bdb24c19f254aa0496d3933920f3ae92f85a0/usbhost.h#L158-L162.

xxxajk, 2017-08-16:
@Lauszus difference--- UHS30 needs an actual real IRQ pin....

shadeyinka, 2017-08-16:
Thanks so much. I will check it out. However in your previous post

I guess you meant,
"Connect pin 53 to pin 3" and not "Connect pin 54 to pin 3".
Pin 54 & Pin 55 are GND in Aduino Mega

xxxajk, 2017-08-16:
If this is a regular Arduino mega with a USB host shield, then the jump from 9 to 3 should work.
The ADK isn't the same thing and this pins are in a different location.

shadeyinka, 2017-08-17:
I think I found out something on the Aduino Mega ADK for use as USB Host

1. The MAX3421E communicate with Arduino with the SPI bus. So it uses the following pins:
Digital: 7 (RST), 50 (MISO), 51 (MOSI), 52 (SCK) 

2. Digital pin 7 should not be used as input or output because is used in the communication with MAX3421E

3. There are three Connections to the MAX3421E through Non broken out on Headers: 
		PJ3 (GP_MAX),               Pin66 on ATMEGA 2560
		PJ6 (INT_MAX),              Pin69,pin8 on ATMEGA 2560
		PH7 (SS).			     Pin27 on ATMEGA 2560			Pin53 on Arduino ADK Header

http://www.electroschematics.com/7965/arduino-mega-adk-pinout/

4. There seems to be an error in USB_HOST_SHIELD.h for the Aduino ADK.
**lines** 
81	  #elif defined(BOARD_MEGA_ADK)
82      #define UHS_MAX3421E_SS_ 53
83      #define UHS_MAX3421E_INT_ 54

54 in line 84 should be referenced to PJ6 (since it has no broken out header, I don't know how to properly label it). *I could be wrong about this Sir

Pin 54 on Aduino Mega and ADK is GND
http://download.arduino.org/products/MEGAADK/arduino-mega-adk-schematic.pdf

5. Is there any other library files that needs to be modified in UHS30 appart from USB_HOST_SHIELD.h to effect this changes.

Thanks

shadeyinka, 2017-08-17:
Hi,
In USB_Host_Shield_2.0: avrpins.h
P54 and P55 (line 514-517) were properly defined. I am not sure of whether it was so done in UHS30

#ifdef BOARD_MEGA_ADK // These pins are not broken out on the Arduino ADK
#define P54 Pe6 // INT on Arduino ADK
#define P55 Pj2 // MAX_RESET on Arduino ADK
#endif

Sorry for my troubling.
Here is my Board
https://www.aliexpress.com/store/product/MEGA-ADK-2560-R3-Enhanced-version-Swith-for-I-O-5V-3-3V-USB-CH340G-USB/1950989_32672086405.html?spm=2114.12010608.0.0.2995e5c0nS6V2R

xxxajk, 2017-08-17:
You will need to locate where the INT_MAX pin is, and connect it to pin 3.
The MEGA ADK has different pin numbers than the regular MEGA.
What I see in your picture is basically a MEGA2560 with the MAX3421E on-board.

Do they provide a schematic?
Perhaps you need to compile for plain MEGA2560?

xxxajk, 2017-08-17:
FYI the official Arduino Mega ADK is totally different from this board too.

xxxajk, 2017-08-17:
Any chance on donating one to me so i can figure it out?

shadeyinka, 2017-08-18:
1. They did not provide a schematic. I have wriien the sellers though, they promised to get back to me.
2. INT_MAX seems to be hardwired to MAX3421E chip...I am still trying to confirm that though.
3. I will attemt to comple of the plain MEGA2560 as suggested.
4. I could donate a board if you send me your shipping address in private. (You know my email address). 
5. In the mean while, can you help me with solving this problem of moving data from SD card to USB Flashdrive with UHS20?(Since I already have a working code with the ADK)..

The working code is:
https://github.com/Microduino/Microduino_Tutorials/tree/master/Microduino_Libraries/_06_Microduino_USBHOST_MAX3421E

Thanks

xxxajk, 2017-08-18:
Definitions seem to be identical, I'll see what I can find out as far as the *actual* tie points for these pins.

xxxajk, 2017-08-18:
```
#ifdef BOARD_MEGA_ADK // These pins are not broken out on the Arduino ADK
#define P54 Pe6 // INT on Arduino ADK
#define P55 Pj2 // MAX_RESET on Arduino ADK
#endif

```

Looks like you will have to find a good place to tap in on these two wires.

Are you capable of doing fine soldering?

xxxajk, 2017-08-18:
P54 I mean, not the reset one :-)

xxxajk, 2017-08-18:
Looks like this is actually a valid /INT pin!

shadeyinka, 2017-08-18:
Thanks. I will check out the Board for the possibility of tapping PE6 and PJ2. I am not sure I can do fine soldering though but I am willing to givebit a try.

Is there no way round UHS20 for the working code code, while solution is being sought for the ADK board?

https://github.com/Microduino/Microduino_Tutorials/tree/master/Microduino_Libraries/_06_Microduino_USBHOST_MAX3421E

Thanks

xxxajk, 2017-08-18:
Well, there are two options to get this to work without soldering.
1: make a new set of read/write/mode for the digital pins.
OR
2: fix the definitions for the ADK in the Arduino definitions....

I think the first one makes the most sense at this point, and is very easy to do.

shadeyinka, 2017-08-18:
1. Which of the libraries need to be tweeked to Making a new set of read/write/mode for the digital pins in  the library and How do you suggest it be implemented.
(I wouldn't mind going by this as a first option before having a permanent and generic solution to the problem-at least to have a working project).
The problem looks so simple, yet daunting.

2. I assume it is more difficult to get the UHS20 move data between SD card and USB FlashDrive.

xxxajk, 2017-08-18:
Easier to work around in UHS30, UHS2 is horrible for this.
Working on the patch for you to test, it seems to only be half broken :-)

shadeyinka, 2017-08-18:
I just got the schematic to the Arduino Mega ADK board by Robotdyn. I will send it now through my mail.

xxxajk, 2017-08-18:
Please send to my personal email, available off of my own github page. Just click my Avatar :-)

shadeyinka, 2017-08-18:
Here is the Schematic for the Mega ADK. I just got it from the suppliers.

<https://github.com/felis/UHS30/issues/32#issuecomment-323290734>, or mute
the thread

xxxajk, 2017-08-18:
Send to my personal email as an attachment please.

shadeyinka, 2017-08-18:
Thanks in anticipation for the patch. A million cheers to you if it works.

Thanks

xxxajk, 2017-08-18:
On your board, which of the following resistors are populated (IF THESE ARE MARKED!)
R11
R12
R13
R16

shadeyinka, 2017-08-18:
I hope you can confirm this from the schematic
MAX3421E.      ATMEGA2560.            ADUINO ADK
RST_3V3            PH4/PJ2.                    PIN7 
SCK_3V3.           PB1.                             PIN52
SS_3V3.              PB0.                             PIN53
MOSI_3V3.         PB2.                             PIN51
MISO_3V3.         PB3.                             PIN50
GPC_3V3.           PH5/PJ3.                    PIN10
INT_3V3.            PH6/PE6.                     PIN9

If this is true, an easy work around should be possible

shadeyinka, 2017-08-18:
Sorry. This is gabbled up. The formating was messed up. I will send it through my mail again

shadeyinka, 2017-08-18:
I just resent it as an *.rtf attachment.

shadeyinka, 2017-08-18:
All the resistors mentioned are marked

xxxajk, 2017-08-18:
Try now...

shadeyinka, 2017-08-18:
What about the following changes
1. In USB_HOST_SHIELD.h 
81	#elif defined(BOARD_MEGA_ADK)
82 #define UHS_MAX3421E_SS_ 53
83 #define UHS_MAX3421E_INT_ 9 //modified from 54 to 9

2. In what other files can one rectify this?

xxxajk, 2017-08-18:
Leave it as 54

shadeyinka, 2017-08-18:
Sorry, I checked a wrong Board. 
None of the components were labelled.

xxxajk, 2017-08-18:
Not a problem. Does the new code work?

shadeyinka, 2017-08-18:
1. I havent seen the New Code. Did you send it to my mail or you uploaded it to your github page? 

2. I hope you have confirmed this from the schematic. They look normal and INT can be taken fro  Pin9, I dont know why it is not working
MAX3421E............... ATMEGA2560............... ADUINO ADK
RST_3V3 __________PH4/PJ2.______________PIN7
SCK_3V3__________PB1___________________PIN52
SS_3V3____________PB0___________________PIN53
MOSI_3V3________PB2___________________PIN51
MISO_3V3________PB3___________________PIN50
GPC_3V3__________PH5/PJ3______________PIN10
INT_3V3___________PH6/PE6______________PIN9

I reverted back line 83 to #define UHS_MAX3421E_INT_ 54

xxxajk, 2017-08-18:
Do a pull from github :-)

shadeyinka, 2017-08-18:
Ok.
I will feed you back as soon as I test it. Thanks

xxxajk, 2017-08-18:
no problem :-)

shadeyinka, 2017-08-18:
The new code still get stuck when executing the line

while(KINETIS_Usb.Init(1000) != 0);

I have put the jumper 
between pin9 and pin3
between pin53 and pin3
with no jumper

xxxajk, 2017-08-18:
Should be no jumpers needed with new code.

xxxajk, 2017-08-18:
Also compile with board as the MEGA ADK

shadeyinka, 2017-08-18:
Even with no jumper.
It prints **Start**. on the serial monitor  and then waits indefinitly

xxxajk, 2017-08-18:
Try this with the board_qc example.

shadeyinka, 2017-08-18:
Yes. I did compile it as
1. Mega ADK
2. Mega 2560
3. Changed Board in case it is faulty
Nothing seem to work

xxxajk, 2017-08-18:
Please try board_qc as Mega ADK, no jumpers, and it should print something out, after you pull this new patch.

shadeyinka, 2017-08-18:
What do you mean by board_qc?
Did you upload a new patch?

xxxajk, 2017-08-18:
https://github.com/felis/UHS30/blob/master/libraries/UHS_host/USB_HOST_SHIELD/examples/board_qc/board_qc.ino
It is under the examples for USB Host Shield.
Yes I uploaded a new patch. :-)

shadeyinka, 2017-08-18:
This is the Result from the Serial Display

Circuits At Home 2011
USB Host Shield Quality Control RoutineInitial reset failed.
Unrecoverable error - test halted!!
0x55 pattern is transmitted via SPI
Press RESET to restart test

[32 more replies]

closed 2018-02-16 by xxxajk