felis/UHS30 #37: malloc()/new failed with AVR.

YuuichiAkagawa, 2018-02-17:
I am testing MIDI driver with MEGA ADK.
malloc() and new fail in the AVR toolchain after version 1.6.12.

- Work: 1.6.2 to 1.6.11
- Not work: 1.6.12 to 1.6.20

However, in version 1.6.11 or earlier, UNO will result in a link error.
```
Temp\arduino_build_328310\sketch\USB_MIDI_dump.ino.cpp.o: In function `MAX3421E_HOST::ISRbottom()':

UHS_host/USB_HOST_SHIELD/USB_HOST_SHIELD_INLINE.h:762:(.text._ZN13MAX3421E_HOST9ISRbottomEv+0x48): relocation truncated to fit: R_AVR_7_PCREL against `no symbol'

collect2.exe: error: ld returned 1 exit status
```

Note: ARM platform is work fine.

xxxajk, 2018-02-17:
I will have to look deeper into this AVR problem. 
UHS provides a replacement heap arena management.

xxxajk, 2018-02-18:
BTW, the link error is a known ld bug.

1.6.9 is known to work fine.