felis/UHS30 #38: F0x01 error

saikiran7714, 2018-07-27:
Sir im using this library for max3421e usb host controller with Arduino mega2560 but when i plug my pendrive I'm getting F0x01 error please help me

xxxajk, 2018-07-27:
did you add the IRQ jumper as per the instructions?

saikiran7714, 2018-07-28:
yes sir i have attached jumper wires as shown in images but getting same error
[image]

xxxajk, 2018-07-28:
try this example:
https://github.com/felis/UHS30/blob/master/libraries/UHS_FS/examples/USB_HOST_SHIELD/UHS_FS_NEW_DEMO/UHS_FS_NEW_DEMO.ino

xxxajk, 2018-07-28:
Also, 0x01 is UHS_USB_HOST_STATE_DEBOUNCE, it isn't an error :-)

saikiran7714, 2018-07-30:
ok  sir but it is stucked at Debounce state  only not going beyond it. So that i cant able to write files to pen drive

xxxajk, 2018-07-30:
How are you powering the mega? USB only, or AC adapter? If USB powering it, try using a 9 volt DC adapter, since your USB port on the computer can't supply enough power. If that does not solve the issue, try a different brand of drive.

saikiran7714, 2018-07-30:
no sir im powering from 12v adapter .  If i upload UHS_FS_NEW_DEMO.ino in serial monitor only character S is printed

saikiran7714, 2018-07-30:
[image]
[image]
and my board look like this please tell me if anything needs to be done

xxxajk, 2018-07-30:
As I suspected, you followed the wrong instructions.
It looks like you did the instructions for PIC32. 
Don't worry, you have not broken your shield. :-)
Connect pin 9 to pin 3 with a jumper wire.

saikiran7714, 2018-07-31:
[image]
thank you very much sir its working . Please share a example about howto creating a .txt file and writing to it and also can i use sd card module with it

xxxajk, 2018-07-31:
How to create and write, read and delete, and even list files is in the demo.
Just like Linux/MacOS/UNIX, there is not difference between text files or any other kind of file.
Just like Linux/MacOS/UNIX, it uses the disk label as the mount point.

More-or-less like what you would find in just about any C or C++ book, file operations are the same, with a few differences. 

Most notably there is no FILE struct. Therefore any example code that you find which uses  will not work out of the box, and will require modifications. for example, if you see something like `FILE *fptr;`, you will need to modify to use file descriptor numbers. The reason for the lack of this is that buffering isn't really possible when there isn't a lot of RAM. It also tends to bloat everything with an additional abstraction layer, reducing available flash, and would actually make things slower.

If you follow any demo in a book, or code in the wild that uses file descriptors, such as `open()` you can just append `fs_` to `open()` like this --> `fs_open()` and most simple examples will work.

This is very standard stuff, so I suggest that you do a little bit of web searching.
Here is a helpful page for you, which contains  [examples](http://codewiki.wikidot.com/system-calls). With very few modifications they should work without problems. Note that not all of the functions on that page are implemented in UHS3, and never will. e.g. `dup` and `dup2`.
One thing that _IS_ allowed in this library is seeking to positions within the file.

The ultimate documentation to the library is the library code. See [this](https://github.com/felis/UHS30/blob/master/libraries/UHS_FS/UHS_FS.h#L68-L97) for a list of functions available. 

Hope that is all helpful information.

saikiran7714, 2018-08-07:
Sir can i use sd card library and uhs30 both at same time . My project is copying data from SD card to pendrive.

saikiran7714, 2018-08-07:
In file included from C:\Users\SAI\Documents\Arduino\libraries\UHS_host/UHS_host.h:56:0,

                 from C:\Users\SAI\Documents\Arduino\sketch_aug07a\sketch_aug07a.ino:106:

C:\Users\SAI\Documents\Arduino\libraries\UHS_host/UHS_KINETIS_FS_HOST/UHS_KINETIS_FS_HOST.h:24:2: error: #error include dyn_swi.h first

 #error include dyn_swi.h first

  ^
Sir im getting this error please help me

xxxajk, 2018-08-07:
**KINETIS is for Freescale/NXP ARM chips**, not Atmel/Microchip AVR with USB Host Shield. **Use the correct define statements in your sketch.**

No, you can't safely mix SD Library and UHS without massive changes and major bug fixes to SD Library. I wonder why anyone would want to do such a thing.
**The SD library has some very nasty bugs** with creating/writing/appending files, and IIRC does not allow in-place changes that work correctly. My implementation does not have any of these problems, and provides a more standard API.

However, there are two solutions that will work, and you should set the open file limit higher if you intend on having more than one open file at the same time. Just define _FS_LOCK to the maximum simultaneous open files you expect to have.

**1: use a USB HUB and USB sdcard reader, with the volume label as SDCARD, or just use two pen drives, and make the volume label SDCARD on one of them. This is the easiest solution.**
2: use two USB Host shields and sdcard reader, with the volume label as SDCARD, or just use two pen drives, and make the volume label SDCARD on one of them. Some trace cutting and soldering to one of the  shields is required.

What allows this to always work is the fact that the volume label becomes the path to access the media, and to detect if the media is present and mounted.You can use any label, but your code must be able to know what to expect. Never name two volumes with the same label, because only the last discovered one will be used, and the prior volumes could become corrupt. Please review the comment in the file system example mentioned 9 days ago for information on how this works.

Vamshi253, 2018-10-04:
Hi Saikiran,

This is Vamshi. At the moment, I am working on a similar project that you were working on. I have some doubts regarding the project. Can you please provide your email so that I can send queries to you. I am requesting you to please help me in this regard.
Thank You.