felis/UHS30 #39: MIDI don't run continues

vinhtrang90, 2018-09-23:
i'm really thank you for this library.
I use it for get data form my keyboard. via USB midi.
I use usb host for arduino, and arduino mega 2560.
when connected, data send not continues, it like a "LAG" for some note, and after few seconds, it tranmist again. 
I don't know why is that. Please help me ! thank you !

P/s: when connect keyboard with PC, this work well.

vinhtrang90, 2018-09-23:
hi YuuichiAkagawa.
 don't know toolchain version, what is this ?!
the IDE ver ? i use 1.6.8
The keyboard is KZM- K200.
Thank you !

vinhtrang90, 2018-09-23:
hi,
i'm test again with serial.print.
i see "Disconnected from MIDI" message.
and don't send any thing.
I try re connect when this disconnect. but have same problem. again...again...

i had use old version library of you, have same problem. but it auto reconnect. version 3 library don't auto reconnect

i tried disable LOAD_UHS_PRINTF_HELPER feature ==> same problem !
Many thank you !

p/s: sorry because my english skill is not good !

vinhtrang90, 2018-09-23:
this is code:
`/*
 *******************************************************************************
   USB-MIDI to Legacy Serial MIDI converter with SysEx support
   Copyright 2017 Yuuichi Akagawa

   based on USBH_MIDI for USB Host Shield 2.0.
   https://github.com/YuuichiAkagawa/USBH_MIDI

   This is sample program. Do not expect perfect behavior.

   Note: This driver supports for MIDI Streaming class only.
         If your MIDI Controller does not work, probably you needs its vendor specific driver. ******************************************************************************
*/

// Patch printf so we can use it.
//#define LOAD_UHS_PRINTF_HELPER
// Load the USB Host System core
#define LOAD_USB_HOST_SYSTEM
// Load USB Host Shield
#define LOAD_USB_HOST_SHIELD
// Load MIDI class driver
#define LOAD_UHS_MIDI

//#define ENABLE_UHS_DEBUGGING 1
//#define DEBUG_PRINTF_EXTRA_HUGE 1
//#define DEBUG_PRINTF_EXTRA_HUGE_UHS_HOST 1
//#define DEBUG_PRINTF_EXTRA_HUGE_USB_HOST_SHIELD 1
//#define DEBUG_PRINTF_EXTRA_HUGE_MIDI_HOST 1
//#define UHS_DEVICE_WINDOWS_USB_SPEC_VIOLATION_DESCRIPTOR_DEVICE 1

// Note: useless check!
// Should really be checking:
// SERIAL_PORT_MONITOR
// SERIAL_PORT_USBVIRTUAL
// SERIAL_PORT_HARDWARE_OPEN
// not USBCON (OR USBCON ONLY) PLEASE FIX THIS!

#ifdef USBCON
#define _MIDI_SERIAL_PORT Serial1
#define USB_HOST_SERIAL Serial
#else
#define _MIDI_SERIAL_PORT Serial
#define USB_HOST_SERIAL Serial1
#endif

#include <Arduino.h>
#ifdef true
#undef true
#endif
#ifdef false
#undef false
#endif

#include <UHS_host.h>

MAX3421E_HOST *UHS_Usb;
UHS_MIDI *Midi;
bool connected;

// Poll USB MIDI Controller and send to serial MIDI

void MIDI_poll() {
  uint8_t size;
  uint8_t outBuf[4];
  uint8_t sysexBuf[3];

  do {
    if ((size = Midi->RecvRawData(outBuf)) > 0) {
      //MIDI Output
      uint8_t rc = Midi->extractSysExData(outBuf, sysexBuf);
      if (rc != 0) { //SysEx
        _MIDI_SERIAL_PORT.write(sysexBuf, rc);
      } else { //SysEx
        _MIDI_SERIAL_PORT.write(outBuf + 1, size);
      }
    }
  } while (size > 0);
}

void setup() {
  connected = false;
  while (!USB_HOST_SERIAL) {
    yield();
  }
  Serial.begin(115200);
  //USB_HOST_SERIAL.begin(115200);
  delay(100);
  UHS_Usb = new MAX3421E_HOST();
  Midi = new UHS_MIDI(UHS_Usb);
  while (UHS_Usb->Init(1000) != 0);
}

void loop() {
  if (Midi->isReady()) {
    if (!connected) {
      connected = true;
      Serial.println("Connected to MIDI");
    }
    MIDI_poll();
  } else {
    if (connected) {
      connected = false;
      Serial.println("Disconnected from MIDI");
      delay(10);
      UHS_Usb = new MAX3421E_HOST();
      Midi = new UHS_MIDI(UHS_Usb);
      while (UHS_Usb->Init(1000) != 0);
    }
  }
}`

this is message print via serial:
 `Connected to MIDI
..................................................................................................................................................................................................................................................................................................................................Disconnected from MIDI`
with "......." is 0xF8 (timing clock) & 0xFE

YuuichiAkagawa, 2018-09-24:
I tested with Arduino Mega ADK and IDE1.6.8(toolchain version 1.6.10).
(MIDI devices is KORG nanoKEY and nanoKontrol)
It seems to work normally.

```
// Patch printf so we can use it.
#define LOAD_UHS_PRINTF_HELPER
// Load the USB Host System core
#define LOAD_USB_HOST_SYSTEM
// Load USB Host Shield
#define LOAD_USB_HOST_SHIELD
// Load MIDI class driver
#define LOAD_UHS_MIDI

#define _MIDI_SERIAL_PORT Serial
#define USB_HOST_SERIAL Serial1

#include <Arduino.h>
#ifdef true
#undef true
#endif
#ifdef false
#undef false
#endif

#include <UHS_host.h>

MAX3421E_HOST *UHS_Usb;
UHS_MIDI *Midi;
bool connected;

// Poll USB MIDI Controller and send to serial MIDI

void MIDI_poll() {
  uint8_t size;
  uint8_t outBuf[4];
  uint8_t sysexBuf[3];

  do {
    if ((size = Midi->RecvRawData(outBuf)) > 0) {
      //MIDI Output
      uint8_t rc = Midi->extractSysExData(outBuf, sysexBuf);
      if (rc != 0) { //SysEx
        _MIDI_SERIAL_PORT.write(sysexBuf, rc);
      } else { //SysEx
        _MIDI_SERIAL_PORT.write(outBuf + 1, size);
      }
    }
  } while (size > 0);
}

void setup() {
  connected = false;
  Serial.begin(115200);
  //USB_HOST_SERIAL.begin(115200);
  delay(100);
  UHS_Usb = new MAX3421E_HOST();
  Midi = new UHS_MIDI(UHS_Usb);
  while (UHS_Usb->Init(1000) != 0);
}

void loop() {
  if (Midi->isReady()) {
    if (!connected) {
      connected = true;
      Serial.println("Connected to MIDI");
    }
    MIDI_poll();
  } else {
    if (connected) {
      connected = false;
      Serial.println("Disconnected from MIDI");
    }
  }
}
```

Check for toolchain version in IDE.
```
Tools->Board: ->Boards Manager...
Arduino AVR Boards Arduino version X.X.XX
```

@xxxajk Do you have Arduino Mega?

vinhtrang90, 2018-09-24:
thank you !
my keyboard not work well. i trying to test again.
Thank you for your help.

my toolchain ver is 1.6.10 . IDE 1.6.8

vinhtrang90, 2018-09-24:
hello, i'm just test with piano electric. it work well.
maybe my keyboard have some problem. this usually send "0xF8" and "0xFE" any time.
and when press key, the message is long. can make MCU "lag". "not enough speed to process data" i think.

or maybe not comfortable with my keyboard.... so sad !

YuuichiAkagawa, 2018-09-24:
Thank you for your report.

These are system real time messages.
0xF8 : Timing Clock (sends 24 clocks per quarter note)
0xFE : Active Sensing(send a status byte every 300 ms)

Can your keyboard disable these feature?

vinhtrang90, 2018-09-25:
hi,
"Can your keyboard disable these feature?" ==> NO, i trying to turn off it, but i can't. 

can you tell me, or help me to check the time "USB host request data from device" ?! i don't see ( or don't full understand) lib. how can change the frequency request data ?!
thank you !

YuuichiAkagawa, 2018-09-26:
I tested the timing clock receive. 
No lag was seen.

Connection:
iPad(Cubasis) -> Yamaha i-MX1 --MIDI--> Roland DUO-CAPTURE EX --USB--> Arduino Mega ADK --USB Serial--> PC(SerialDebugTool)

Movie:
https://youtu.be/uBx_rK5nB1w

> can you tell me, or help me to check the time "USB host request data from device" 

UHS30 is interrupt driven.

0. Registration of the interrupt handler
https://github.com/felis/UHS30/blob/master/libraries/UHS_host/USB_HOST_SHIELD/USB_HOST_SHIELD_INLINE.h#L379
https://github.com/felis/UHS30/blob/master/libraries/UHS_host/USB_HOST_SHIELD/USB_HOST_SHIELD.h#L230
https://github.com/felis/UHS30/blob/master/libraries/UHS_host/USB_HOST_SHIELD/USB_HOST_SHIELD.h#L236

1. Interrupt from MAX3421E
https://github.com/felis/UHS30/blob/master/libraries/UHS_host/USB_HOST_SHIELD/USB_HOST_SHIELD_INLINE.h#L41-L42
https://github.com/felis/UHS30/blob/master/libraries/UHS_host/USB_HOST_SHIELD/USB_HOST_SHIELD_INLINE.h#L862
https://github.com/felis/UHS30/blob/master/libraries/UHS_host/USB_HOST_SHIELD/USB_HOST_SHIELD_INLINE.h#L945
https://github.com/felis/UHS30/blob/master/libraries/UHS_host/USB_HOST_SHIELD/USB_HOST_SHIELD_INLINE.h#L744
https://github.com/felis/UHS30/blob/master/libraries/UHS_host/USB_HOST_SHIELD/USB_HOST_SHIELD_INLINE.h#L823

2. Polling to MIDI driver
https://github.com/felis/UHS30/blob/master/libraries/UHS_host/UHS_MIDI/UHS_MIDI_INLINE.h#L147
https://github.com/felis/UHS30/blob/master/libraries/UHS_host/UHS_MIDI/UHS_MIDI_INLINE.h#L152

3. Reading data
https://github.com/felis/UHS30/blob/master/libraries/UHS_host/UHS_MIDI/UHS_MIDI_INLINE.h#L134
https://github.com/felis/UHS30/blob/master/libraries/UHS_host/UHS_MIDI/UHS_MIDI_INLINE.h#L137

xxxajk, 2018-09-26:
The actual polling rate is set on line 88.
https://github.com/felis/UHS30/blob/master/libraries/UHS_host/UHS_MIDI/UHS_MIDI_INLINE.h#L84-L88
I do not recommend anything lower than 2, and the value should never be less than 1.

vinhtrang90, 2018-09-30:
HI !
FYI !
i trying use USB host to build this : https://youtu.be/fQoMeIYgI9g

works well with electric piano. (KZM-K200 not good, i use my pc to though midi data from keyboard to USB nagative on arduino micro)

thank you !
enjoy it: https://youtu.be/wOqSsIkrY9A