felis/UHS30 #40: Not working on ftdi chip

theMASTERMINDpK, 2018-11-24:
I'm trying to connect USB host shield with the device having FTDI Chip to provide virtual com port. But failed to build a connection between host and device. I'm using USB host library 2.0 available on GitHub.
Does anyone knows how to Implement FTDI on this library??
Some device specs are listed below:
VIRTUAL COM PORT
Baud Rate:9600 baud* Format: 8 data bits, no parity, 1 stop bit*
Protocol: Modbus over Serial Line

xxxajk, 2018-11-24:
Ftdi is supported in UHS 3.0. it is emulating CDC ACM with extensions.
Try the CDC ACM example code.
If the vendor ID or device ID is not recognized and it really is an ftdi
part you will need to supply that information to me so that I can support
it.

theMASTERMINDpK, 2018-12-05:
Hi thank you for help. I tried that example but failed. My device spec are as follows:
Description: USB Serial Converter(FTDI FT230X Basic UART)
Device Type: Vendor Specific

Protocal Version: USB 2.00
Current Speed: Full Speed
Max Current: 90mA

USB Device ID: VID = 0403 PID = 6015
Serial Number: DO2IZ7XE

Device Vendor: FTDI
Device Name: FT230X Basic UART
Device Revision: 1000

Controller Part-Number: Unknown

xxxajk, 2018-12-06:
Please turn on debugging and post the entire output.
This chip may have a quirk I need to do a fix for.
I will have to see if I have this chip in my FTDI collection...

theMASTERMINDpK, 2018-12-06:
When i compile the code each example gives me the error:
Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Uno"

In file included from C:\Users\Yasir Shahzad\Documents\Arduino\libraries\UHS_host/UHS_host.h:48:0,

                 from C:\Users\Yasir Shahzad\Documents\Arduino\libraries\UHS_host\UHS_CDC_ACM\examples\USB_HOST_SHIELD\acm_terminal\acm_terminal.ino:35:

C:\Users\Yasir Shahzad\Documents\Arduino\libraries\UHS_host/UHS_host_INLINE.h: In member function 'uint8_t UHS_USB_HOST_BASE::Configuring(uint8_t, uint8_t, uint8_t)':

C:\Users\Yasir Shahzad\Documents\Arduino\libraries\UHS_host/UHS_host_INLINE.h:267:30: error: request for member 'maxPktSize' in 'p->UHS_Device::epinfo[0]', which is of pointer type 'volatile UHS_EpInfo*' (maybe you meant to use '->' ?)

                 p->epinfo[0].maxPktSize = 0x08; // USB Spec, start small, work your way up.

                              ^

exit status 1
Error compiling for board Arduino/Genuino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

xxxajk, 2018-12-06:
Bug fixed.
Please use an Arduino Mega.
The Arduino Uno does not have enough memory to run this library at this time, but MIGHT in the future.

theMASTERMINDpK, 2018-12-06:
thanku i'm going to arrange arduino mega and will try. I'm trying This FTDI chip with UHS 2.0 and got some success in getting response after changing the PID value according to sensor in the libarary
[image]

Everything is working fine. Return codes showing success but the output is not correct. I'm sending 8 bytes of ch array  with rcode=0vand should get 7 bytes of data but getting"0x01 0x60 0x60"
Here is example code given in UHS2.0:

 "#include <cdcftdi.h>
#include <usbhub.h>

#include "pgmstrings.h"

// Satisfy the IDE, which needs to see the include statment in the ino too
#include <SPI.h>

class FTDIAsync : public FTDIAsyncOper
{
public:
    uint8_t OnInit(FTDI *pftdi);
};

uint8_t FTDIAsync::OnInit(FTDI *pftdi)
{
    uint8_t rcode = 0;

    rcode = pftdi->SetBaudRate(9600);

    if (rcode)
    {
        ErrorMessage<uint8_t>(PSTR("SetBaudRate"), rcode);
        return rcode;
    }
    rcode = pftdi->SetFlowControl(FTDI_SIO_DISABLE_FLOW_CTRL);

    if (rcode)
        ErrorMessage<uint8_t>(PSTR("SetFlowControl"), rcode);

    return rcode;
}

USB              Usb;
//USBHub         Hub(&Usb);
FTDIAsync        FtdiAsync;
FTDI             Ftdi(&Usb, &FtdiAsync);

void setup()
{
  Serial.begin( 9600 );
#if !defined(__MIPSEL__)
  while (!Serial); // Wait for serial port to connect - used on Leonardo, Teensy and other boards with built-in USB CDC serial connection
#endif
  Serial.println("Start");

  if (Usb.Init() == -1)
      Serial.println("OSC did not start.");

  delay( 200 );
}

void loop()
{
    Usb.Task();
    //Serial.println("loop");
    if( Usb.getUsbTaskState() == USB_STATE_RUNNING )
    {
        delay(2000);
        uint8_t  rcode;
        char cmdbuf[8];
  
  cmdbuf[0] = 0x01;
  cmdbuf[1] = 0x03;
  cmdbuf[2] = 0x00;
  cmdbuf[3] = 0x0B;
  cmdbuf[4] = 0x05;
  cmdbuf[5] = 0x00;
  cmdbuf[6] = 0xF5;
  cmdbuf[7] = 0xC8;
        //char strbuf[]; = {01,03,00,0B,00,01,F5,C8};
        //char strbuf[] = "The quick brown fox jumps over the lazy dog";
        //char strbuf[] = "This string contains 61 character to demonstrate FTDI buffers"; //add one symbol to it to see some garbage
        //Serial.print(".");

        rcode = Ftdi.SndData(strlen(cmdbuf), (uint8_t*)cmdbuf);
       // Serial.print("rcode:");
       //   Serial.println(rcode);

	if (rcode)
            ErrorMessage<uint8_t>(PSTR("SndData"), rcode);

        delay(50);

        uint8_t  buf[64];

        for (uint8_t i=0; i<64; i++)
            buf[i] = 0;

        uint16_t rcvd = 64;
        rcode = Ftdi.RcvData(&rcvd, buf);

        if (rcode && rcode != hrNAK)
            ErrorMessage<uint8_t>(PSTR("Ret"), rcode);

        // The device reserves the first two bytes of data
        //   to contain the current values of the modem and line status registers.
        if (rcvd > 1)
           // Serial.print("DataReceived:");
           // Serial.println();
            Serial.print((char*)(buf+0));
            Serial.print((char*)(buf+1));
            Serial.print((char*)(buf+2));
            Serial.print((char*)(buf+3));
            Serial.print((char*)(buf+4));
            Serial.print((char*)(buf+5));
            Serial.print((char*)(buf+6));
            Serial.print((char*)(buf+7));
       

        delay(10);
    }
}

xxxajk, 2018-12-06:
I am aware of the cross post. :-)

theMASTERMINDpK, 2018-12-06:
Hi I tried using both examples using arduino Meaga2560.After turning on debugging on acm_terminal.ino i'm getting nothing. and on buffered_acm_terminal.ino getting this one as output  **printf("\r\n\r\nWaiting for Connection...\r\n");**. I don't know what to do. Can you give me 20minutes on team-viewer to get it run??  In return I can help you in hardware designing, professional PCB designing, OrangePi raspberryPi coding & will never forget you in my life :)
Here is my email address to contact, thanks
Yasir.shahzad@kics.edu.pk

xxxajk, 2018-12-07:
Have you read the instructions for AVR?
Specifically: Connect pin 9 to pin 3 with a jumper wire.
Let me know if it works after that...

theMASTERMINDpK, 2018-12-09:
I tried the sensor with arudino mega 2560 by connecting pin 9 to pin 3 with a jumper wire. And received the debug messages. At last temperature sensor is connected 👍 . And I don't know how to send and receive bytes in this example. 
Here is a video of my temp sesnsor working with serial terminal:
https://www.youtube.com/watch?v=B1S0dRWjoZA&feature=youtu.be

Here is the example which i uploaded on arduino Mega
https://github.com/felis/UHS30/blob/master/libraries/UHS_host/UHS_CDC_ACM/examples/USB_HOST_SHIELD/acm_terminal/acm_terminal.ino

And received the following debug messages
ctrlReqRePass SPI speed 25000000
Waiting for Connection... 
UHS_USB_HOST_STATE_DEBOUNCE
BEFORE CDIRQ T BEIRQ T resetting F state 0x02
AFTER CDIRQ T BEIRQ T resetting F state 0x02
 UHS_USB_HOST_STATE_DEBOUNCE 
UHS_USB_HOST_STATE_DEBOUNCE_NOT_COMPLETE 
UHS_USB_HOST_STATE_RESET_DEVICE 
UHS_USB_HOST_STATE_RESET_NOT_COMPLETE 
UHS_USB_H
BEFORE CDIRQ T BEIRQ T resetting F state 0x03
AFTER CDIRQ T ST_STATE_RESET_NOT_COMPLETE 
UHS_USB_HOST_STATE_WAIT_BUS_READY

Configuring: parent = 0, port = 1, speed = 1

Configuring PktSize 0x40,  rcode: 0x00, retries 0, 
ep entry for interface 0 ep 0 max packet size = 64 
ctrlReq2: left: 64, read:64, nbytes 64 
ctrlReqRead left: 64 
Requesting 64 bytes Got 8 bytes 
ctrlReqRead left: 56, read 8 
ctrlReq3: acceptBuffer sz 8 nbytes 64 left 56

BEFORE CDIRQ T BEIRQ T resetting T state 0x0e

AFTER CDIRQ T BEIRQ T resetting T state 0x0e 
ep entry for interface 0 ep 0 max packet size = 64 
0 retries. 
DevDescr 2nd poll, bMaxPacketSize0:8 
ep entry for interface 0 ep 0 max packet size = 8 
ctrlReq2: left: 18, read:18, nbytes 18 
ctrlReqRead left: 18 
Requesting 18 bytes Got 8 bytes 
Got 8 bytes 
Got 2 bytes 
ctrlReqRead left: 0, read 18

BEFORE CDIRQ T BEIRQ T resetting T state 0x0e

AFTER CDIRQ T BEIRQ T resetting T state 0x0e 
ep entry for interface 0 ep 0 max packet size = 64
0 retries. 
configs: 1 
ep entry for interface 0 ep 0 max packet size = 64 
ctrlReq2: left: 9, read:9, nbytes 9 
ctrlReqRead left: 9 
Requesting 9 bytes Got 8 bytes 
ctrlReqRead left: 1, read 8 
ctrlReq2: left: 1, read:9, nbytes 9 
ctrlReqRead left: 1 
Requesting 9 bytes Got 1 bytes 
ctrlReqRead left: 0, read 1 
CONFIGURATION: 0, bNumInterfaces 1, wTotalLength 32 
ep entry for interface 0 ep 0 max packet size = 64 
ctrlReqRead left: 32 
Requesting 8 bytes Got 8 bytes 
ctrlReqRead left: 24, read 8 
bLength: 9 bDescriptorType: 02 
eating 8 
ctrlReqRead left: 24 
Requesting 8 bytes Got 8 bytes 
ctrlReqRead left: 16, read 8 
bLength: 9 bDescriptorType: 04 
INTERFACE DESCRIPTOR FOUND 
ctrlReqRead left: 16 
Requesting 8 bytes Got 8 bytes 
ctrlReqRead left: 8, read 8 
Getting 2 endpoints 
bLength: 7 bDescriptorType: 05 
ENDPOINT DESCRIPTOR: 0 
ctrlReqRead left: 8 
Requesting 8 bytes Got 8 bytes 
ctrlReqRead left: 0, read 8 
ENDPOINT DESCRIPTOR OK 
eating 0 
bLength: 7 bDescriptorType: 05 
ENDPOINT DESCRIPTOR: 1 
ENDPOINT DESCRIPTOR OK 
eating 0 
eating 1 
ENDPOINT DESCRIPTORS FILLED 
TestInterface VID:0403 PID:6015 Class:00 Subclass:00 Protocol 00 
Interface data: Class:ff Subclass:ff Protocol ff, number of endpoints 2 
Parent: 00, bAddress: 01 
Driver 1 supports this interface 
Driver 1 can be used for this interface 
USB_INTERFACE END OF STREAM 
ep entry for interface 0 ep 0 max packet size = 8 
ctrlReq2: left: 9, read:9, nbytes 9 
ctrlReqRead left: 9 
Requesting 9 bytes Got 8 bytes 
Got 1 bytes 
ctrlReqRead left: 0, read 9 
CONFIGURATION: 1, bNumInterfaces 1, wTotalLength 32 
Best configuration is 1, enumerating interfaces. 
ep entry for interface 0 ep 0 max packet size = 8 
ctrlReqRead left: 32 
Requesting 8 bytes Got 8 bytes 
ctrlReqRead left: 24, read 8 
bLength: 9 bDescriptorType: 02 
eating 8 
ctrlReqRead left: 24 
Requesting 8 bytes Got 8 bytes 
ctrlReqRead left: 16, read 8 
bLength: 9 bDescriptorType: 04 
INTERFACE DESCRIPTOR FOUND 
ctrlReqRead left: 16 
Requesting 8 bytes Got 8 bytes 
ctrlReqRead left: 8, read 8 
Getting 2 endpoints 
bLength: 7 bDescriptorType: 05 
ENDPOINT DESCRIPTOR: 0 
ctrlReqRead left: 8 
Requesting 8 bytes Got 8 bytes 
ctrlReqRead left: 0, read 8 
ENDPOINT DESCRIPTOR OK 
eating 0 
bLength: 7 bDescriptorType: 05 
ENDPOINT DESCRIPTOR: 1 
ENDPOINT DESCRIPTOR OK 
eating 0 
eating 1 
ENDPOINT DESCRIPTORS FILLED 
AttemptConfig: parent = 0, port = 1 
Driver 1 supports this interface 
Interface Configured 
ep entry for interface 0 ep 0 max packet size = 8 
Driver 0 @ 00 Skipped 
Driver 1 @ 01 Initialize 
ep entry for interface 0 ep 0 max packet size = 8 
ep entry for interface 0 ep 0 max packet size = 8
 ep entry for interface 0 ep 0 max packet size = 8 
ep entry for interface 0 ep 0 max packet size = 8

Connected.
Total endpoints = (3)3 

Driver 2 no driver at this index. 
Driver 3 no driver at this index. 
Driver 4 no driver at this index. 
Driver 5 no driver at this index. 
Driver 6 no driver at this index. 
Driver 7 no driver at this index. 
Driver 8 no driver at this index. 
Driver 9 no driver at this index. 
Driver 10 no driver at this index. 
Driver 11 no driver at this index. 
Driver 12 no driver at this index. 
Driver 13 no driver at this index. 
Driver 14 no driver at this index. 
Driver 15 no driver at this index. 
ep entry for interface 0 ep 1 max packet size = 64 
Requesting 64 bytes Got 2 bytes 
ep entry for interface 0 ep 1 max packet size = 64 
Requesting 64 bytes Got 2 bytes 
ep entry for interface 0 ep 1 max packet size = 64 
Requesting 64 bytes Got 2 bytes 
ep entry for interface 0 ep 1 max packet size = 64 
Requesting 64 bytes Got 2 bytes

theMASTERMINDpK, 2018-12-10:
Hi andrew, I think this library is good for this sensor. Please see the video and help me how to get response using this library, thanks

xxxajk, 2018-12-11:
So you want to do this manually like the program you show in the video? or automated?

theMASTERMINDpK, 2018-12-11:
I want it automated. Using your Liberary

theMASTERMINDpK, 2018-12-11:
Mean how to send that 8 byte array to sensor and receive response

xxxajk, 2018-12-11:
You would do it much the same way as you do in the other attempt that you wrote.

1: Send the array
2: Read 1 byte at a time until you fill a new array, the driver has the capability to do this in the background. See the buffered demo for how to do that. IMPORTANT! You need to enable and disable polling while accessing the buffers. See the comments in the demo. In your case, wait for 8 or more bytes.
3: do whatever processing you want to do with the returned data.

theMASTERMINDpK, 2018-12-13:
Hi I tried this code but getting the same output. Can you Please Edit this code according to my  need as i'm not good in C++.  i need  7 bytes replied by the sensor.
Here is my code

`
#define CLIENT_SERIAL_SPEED 9600
#define HOST_SERIAL_SPEED 19200

// Load the USB Host System core
#define LOAD_USB_HOST_SYSTEM
// Load USB Host Shield
#define LOAD_USB_HOST_SHIELD
// Use USB hub
#define LOAD_UHS_HUB

// Patch printf so we can use it.

#define LOAD_UHS_PRINTF_HELPER 1
#define ENABLE_UHS_DEBUGGING 1
#define DEBUG_PRINTF_EXTRA_HUGE 1
#define DEBUG_PRINTF_EXTRA_HUGE_UHS_HOST 1
#define DEBUG_PRINTF_EXTRA_HUGE_USB_HOST_SHIELD 1

// These all get combined under UHS_CDC_ACM multiplexer.
// Each should only add a trivial amount of code.
// XR21B1411 can run in a pure CDC-ACM mode, as can PROLIFIC.
// FTDI has a large code and data footprint. Avoid this chip if you can.
#define LOAD_UHS_CDC_ACM
//#define LOAD_UHS_CDC_ACM_XR21B1411
// This needs testing.
//#define LOAD_UHS_CDC_ACM_PROLIFIC
// This needs testing.
#define LOAD_UHS_CDC_ACM_FTDI

#include <Arduino.h>
#ifdef true
#undef true
#endif
#ifdef false
#undef false
#endif

#include <UHS_host.h>

#define RD_BUF_MAX 1024
#define WR_BUF_MAX 1024

MAX3421E_HOST UHS_Usb;
UHS_USBHub hub_1(&UHS_Usb);

class MY_ACM : public UHS_CDC_ACM {
public:
        uint8_t wrbuf[WR_BUF_MAX];
        uint8_t rdbuf[RD_BUF_MAX];
        volatile uint16_t wr_head;
        volatile uint16_t wr_tail;
        volatile uint16_t rd_head;
        volatile uint16_t rd_tail;
        volatile uint8_t lstat;

        
        MY_ACM(UHS_USB_HOST_BASE *p) : UHS_CDC_ACM(p) {
                wr_head = 0;
                wr_tail = 0;
                rd_head = 0;
                rd_tail = 0;
                lstat = 0;
        };
       

        uint8_t OnStart(void);
[... 419 lines cut ...]

Total endpoints = (3)3
Driver 2 no driver at this index.
Driver 3 no driver at this index.
Driver 4 no driver at this index.
Driver 5 no driver at this index.
Driver 6 no driver at this index.
Driver 7 no driver at this index.
Driver 8 no driver at this index.
Driver 9 no driver at this index.
Driver 10 no driver at this index.
Driver 11 no driver at this index.
Driver 12 no driver at this index.
Driver 13 no driver at this index.
Driver 14 no driver at this index.
Driver 15 no driver at this index.
ep entry for interface 0 ep 1 max packet size = 64
Requesting 64 bytes Got 2 bytes 
ep entry for interface 0 ep 1 max packet size = 64
Requesting 64 bytes Got 2 bytes 
ep entry for interface 0 ep 1 max packet size = 64
Requesting 64 bytes Got 2 bytes 
ep entry for interface 0 ep 1 max packet size = 64
Requesting 64 bytes Got 2 bytes 
ep entry for interface 0 ep 1 max packet size = 64

xxxajk, 2018-12-14:
all you need to do is... write your data, then wait... e.g. 

`
while(Bcm->rd_available() < how_many_bytes_you_want) yield(); // wait
`
then read your data, and do whatever you need to do with that data.

theMASTERMINDpK, 2018-12-14:
Hi andrew thank u for great Help. It worked. if you need any favor regarding Hardware design or PCB design Please do consult me. I will always be available for you, thanks

closed 2018-12-14 by theMASTERMINDpK