felis/UHS30 #44: Migrating Marlin to UHS3.0. A few questions...

marcio-ao, 2019-05-20:
Hello,

Earlier last year, I incorporated UHS2.0 into Marlin. This year I am investigating migrating Marlin to UHS3.0, since we are seeing issues with one of our boards with the old code and I rather put our resources into fixing issues in UHS3.0 rather than UHS2.0. I believe having us chip in development work would be helpful to everyone!

In doing the preliminary work to intergrate UHS3.0, I notice a couple things:

- The code is much cleaner and better organized (this is good!)
- The license is GPL v2, rather than "GPL v3" or "GPL v2 or any later version" (this is a showstopper!)
- When I try compiling with "USB_HOST_SHIELD_USE_ISR 0", I get an error saying this isn't supported (this could present problems with Marlin).

Marlin makes heavy use of interrupts itself and I feel there is much less risk involved if the USB code can run in polled mode. I notice that in "USB_HOST_SHIELD.h" if ARDUINO_AVR_BALANDUINO is defined, it asks the user to use USB_HOST_SHIELD_USE_ISR as 0. So this suggests that at least some work has been done on the feature.

So, a few questions:

1) Could this code be relicensed to "GPL v2 or any later version", so that I can integrate it into Marlin?  (which is GPL v3).
2) UHS2.0 worked without an interrupt, how difficult would it be to make UHS3.0 work without it too?
3) How much work has actually been done in working without interrupts? What is the ARDUINO_AVR_BALANDUINO and how does that work without interrupts?

Any help with these questions are very much appreciated! Thank you!

-- Marcio

xxxajk, 2019-05-20:
1: license will be either GPL3 or GPL3 + BSD, I just have not gotten to changing that. Don't worry, go for it now if you like. I'll try to push that in the next few days.
2: IRQ is now mandatory. Working without IRQ causes USB timing problems, the IRQ now provides for a fully independent timer for events. Yes, some times the IRQ can "block" the regular code from running, but all other IRQ, etc in-fact run. The same would happen in loop() anyway. If you got critical timed code, you should be running it in an IRQ on a timer anyway.
3: That was/is a personal project by @Lauszus Don't know if he tried it.

Please contact me directly for more details and help via my Gmail address.

marcio-ao, 2019-05-20:
If there is anything I can do to help move these things along, please let me know. For example, if you would like, I can make a PR with the license change to header files, then all you need to do is approve it and merge it.

You can check my github page and see that we contribute a lot of time to Marlin -- I wouldn't mind doing the same here if it's code we can use in our printers :)

> Working without IRQ causes USB timing problems, the IRQ now provides for a fully independent timer for events.

For what it is worth, the previous library seems to work fine enough in Marlin for printing files, even in polling mode. We're not talking about high bandwidth needs with 3D printing and occasional errors are tolerated. Marlin has it's own HAL for all digital IO and interrupts. If it's only SPI, CS and INTR, its a fairly trivial to hook in--if interrupts are involved, it could be harder (or not, honestly, I won't know until I try!)

> Please contact me directly for more details and help via my Gmail address.

Is that your preferred means of communicating? If you don't have a preference, I like keeping discussions on Github, in case anyone else could benefit from them.

xxxajk, 2019-05-20:
For timing, it isn't about bandwidth. It is more about when to talk to the physical USB. It also allows the timing to be independent of any built-in timers, as so it won't miss important events. This had always been a major problem on UHS2 for various bits, causing the need to do all sorts of strange things to get something to work. The major benefit is that it eliminates having to worry about doing any of the work that is normally done in the background... you just use it, and it works, much like serial, not having to worry about sending each byte, you just do it in the code, and it happens.

You won't have to manage IRQ or pin defs as you think. This is all taken into account for within UHS3. Try a couple of the examples to get an idea on how things flow much better. 

I also took a look at the marlin code and noticed that you use SDfat... which is buggy as heck... and UHS3 already supports SDcards due to popular demand. :-)

Yeah, I'll get the messages quicker, for one. I also have some out-of-scope questions, etc.

marcio-ao, 2019-05-20:
> I also took a look at the marlin code and noticed that you use SDfat... which is buggy as heck... and UHS3 already supports SDcards due to popular demand. :-)

Yes, the Marlin SD stuff is a mess and its horrible. I agree!

> This had always been a major problem on UHS2 for various bits, causing the need to do all sorts of strange things to get something to work.

Makes sense. I guess the only place where I could see things getting complicated is if UHS3 uses timer interrupts, as Marlin already makes extensive use of those.

> You won't have to manage IRQ or pin defs as you think. This is all taken into account for within UHS3. Try a couple of the examples to get an idea on how things flow much better.

The problem I see is that UHS3 is using Arduino libraries for abstraction. Most 3D printer boards are not Arduino compatible, at least not trivially, and Marlin has its own manner of numbering pins and accessing pins. digitalWrite and digitalRead do not generally work as you would think.

> Yeah, I'll get the messages quicker, for one. I also have some out-of-scope questions, etc.

I'll drop a message to your e-mail. You'll have my e-mail address and can ask your out-of-scope questions then :)

marcio-ao, 2019-05-28:
Closing this ticket since the license change has been merged. Thank you!

closed 2019-05-28 by marcio-ao

xxxajk, 2019-05-28:
No problem!