felis/UHS30 #52: Twitching Servos when using UHS3

marcio-ao, 2019-06-20:
I've found that running an Arduino sketch with UHS3 and the Arduino Servo library causes the servo to twitch. This is one of the problems I am encountering in integrating UHS3 into our printers.

However, since this isolates the problem better, limiting it to the interaction of two libraries, I figured I would pass the information along.

PaulStoffregen, 2019-06-20:
Which board are you using?  One of the old 8 bit AVRs maybe?

xxxajk, 2019-06-20:
Yeah, it is. I'm helping with lulzbot dev paul. :-)

marcio-ao, 2019-06-21:
> Which board are you using? One of the old 8 bit AVRs maybe?

The twitching happens on both an 8-bit AVR as well as a 32-bit ATSAM3X8E. 

Andrew has been helping me off-line, but I just figured I would put this out here for the record.