felis/UHS30 #53: CH340/341a?

foxabilo, 2019-07-12:
Hi all, I was looking to use the library as a starting point to getting the CH340/341a serial chip as one of the supported serial coms devices, anyone know if any of the already done ftdi or prolific would be a good jumping off point? I have been sniffing the usb packets on wireshark and it all looks fairly tame.

xxxajk, 2019-07-13:
Support is planned. I will possibly be doing that this weekend, nice timing :-)

foxabilo, 2019-07-13:
Awesome!

foxabilo, 2019-07-13:
If it is of any use to you, I grabbed an SKR 1.3 controller board talking to Cura with Wireshark from plugging it into the USB socket to Cura sending M105 commands and receiving bulk data. and then closing Cura and unplugging it, basically a full negotiation, enumeration and data cycle. 
[ch340-250000.zip](https://github.com/felis/UHS30/files/3388765/ch340-250000.zip)

foxabilo, 2019-07-13:
And there is this to go from 

https://elixir.bootlin.com/linux/latest/source/drivers/usb/serial/ch341.c

xxxajk, 2019-07-15:
Don't need any infos, but thx for the effort.
I wrote an Android driver for this, and can write it.
I'm also a very old Linux Kernel hacker from the 90's...

foxabilo, 2019-07-15:
Ohhh, great news, I know the sort, used to work with a guy at a games company who could code a driver for a new sensor chip in assembler in a couple of hours with only a data sheet and a pencil. I'm going to have a go myself, just for the challenge ;)

closed 2019-07-16 by foxabilo

mrv96, 2021-01-02:
@xxxajk did you do it at the end? I'm not able to find any reference in the source code