felis/UHS30 #54: Fail SPI speed in acm_terminal

TDuxis, 2019-07-17:
Hi

I am trying to run acm_terminal together with a Huawei USB modem. However, I am getting the following output in the serial window.

[image]

I modified the Setup a bit, as follows

`void setup() {
        Serial.begin( 115200 );
        delay(1000);
        Serial.print("Start\r\n");
        while(!USB_HOST_SERIAL);
        USB_HOST_SERIAL.begin(115200);
        Serial.print("USB HOST SERIAL True\r\n");
        
        printf_P(PSTR("\r\n\r\nStarting CDC-ACM test program...\r\n"));
        while(UHS_Usb.Init(1000) != 0);
        printf_P(PSTR("\r\n\r\nWaiting for Connection...\r\n"));
}`

What may be causing this issue?

Also, as you can see there seems to be some issue with the printf_P function, as I am not getting that output. 

I have included all the defines in the code.
`// Patch printf so we can use it.
#define LOAD_UHS_PRINTF_HELPER 1
#define ENABLE_UHS_DEBUGGING 1
#define DEBUG_PRINTF_EXTRA_HUGE 1
#define DEBUG_PRINTF_EXTRA_HUGE_UHS_HOST 1
#define DEBUG_PRINTF_EXTRA_HUGE_USB_HOST_SHIELD 1`

What could be the issue here?

Thanks, Thomas

xxxajk, 2019-07-18:
Don't mix serial initializations. 
That is what is partially causing the problem.
The other part could be the version of the board support installed, and needs to be sorted yet.

In the boards manager, try downgrading (or upgrading) Arduino AVR Boards by Arduino to version 1.6.11, and see if that helps.

TDuxis, 2019-07-18:
Finally, something :)

I get this. "No driver for this interface." Trafe ends with "Error 0xd1". Is it interface class (=FF) that is spooking again? How can i override that?

`
Pass SPI speed 25000000

Waiting for Connection...
UHS_USB_HOST_STATE_DEBOUNCE

BEFORE CDIRQ T BEIRQ T resetting F state 0x02

AFTER CDIRQ T BEIRQ T resetting F state 0x02
UHS_USB_HOST_STATE_DEBOUNCE
UHS_USB_HOST_STATE_DEBOUNCE_NOT_COMPLETE
UHS_USB_HOST_STATE_RESET_DEVICE
UHS_USB_HOST_STATE_RESET_NOT_COMPLETE
UHS_USB_H
BEFORE CDIRQ T BEIRQ T resetting F state 0x03

AFTER CDIRQ T sfer: addr: 0xc00 ep: 0x574 nbytes: 0x4e00 data: 0x0x21
UHS_USB_HOST_STATE_WAIT_BUS_READY

Configuring: parent = 0, port = 1, speed = 1

Configuring PktSize 0x40,  rcode: 0x00, retries 0,
ep entry for interface 0 ep 0 max packet size = 64
ctrlReq2: left: 64, read:64, nbytes 64
ctrlReqRead left: 64
Requesting 64 bytes Got 18 bytes 
ctrlReqRead left: 46, read 18
RESULT: 0x00 0x00 0x12 0x0040000001000680
ctrlReq3: acceptBuffer sz 18 nbytes 64 left 46

BEFORE CDIRQ T BEIRQ T resetting T state 0x0e

AFTER CDIRQ T BEIRQ T resetting T state 0x0e
ep entry for interface 0 ep 0 max packet size = 64
0 retries.
DevDescr 2nd poll, bMaxPacketSize0:64
ep entry for interface 0 ep 0 max packet size = 64
ctrlReq2: left: 18, read:18, nbytes 18
ctrlReqRead left: 18
Requesting 18 bytes Got 18 bytes 
ctrlReqRead left: 0, read 18
RESULT: 0x00 0x01 0x12 0x0012000001000680

BEFORE CDIRQ T BEIRQ T resetting T state 0x0e

AFTER CDIRQ T BEIRQ T resetting T state 0x0e
ep entry for interface 0 ep 0 max packet size = 64
0 retries.
configs: 1
ep entry for interface 0 ep 0 max packet size = 64
ctrlReq2: left: 9, read:9, nbytes 9
ctrlReqRead left: 9
Requesting 9 bytes Got 9 bytes 
ctrlReqRead left: 0, read 9
RESULT: 0x00 0x01 0x09 0x0009000002000680
CONFIGURATION: 0, bNumInterfaces 1, wTotalLength 32
ep entry for interface 0 ep 0 max packet size = 64
ctrlReqRead left: 32
Requesting 32 bytes Got 32 bytes 
ctrlReqRead left: 0, read 32
bLength: 9 bDescriptorType: 02
eating 8
bLength: 9 bDescriptorType: 04
INTERFACE DESCRIPTOR FOUND
Getting 2 endpoints
bLength: 7 bDescriptorType: 05
ENDPOINT DESCRIPTOR: 0
ENDPOINT DESCRIPTOR OK
eating 0
bLength: 7 bDescriptorType: 05
ENDPOINT DESCRIPTOR: 1
ENDPOINT DESCRIPTOR OK
eating 0
eating 1
ENDPOINT DESCRIPTORS FILLED
TestInterface VID:12d1 PID:1506 Class:00 Subclass:00 Protocol 00
Interface data: Class:ff Subclass:01 Protocol 01, number of endpoints 2
Parent: 00, bAddress: 01
No driver at index 2
No driver at index 3
No driver at index 4
No driver at index 5
No driver at index 6
No driver at index 7
No driver at index 8
No driver at index 9
No driver at index 10
No driver at index 11
No driver at index 12
No driver at index 13
No driver at index 14
No driver at index 15
No driver for this interface.
USB_INTERFACE END OF STREAM
Error 0xd1
`

xxxajk, 2019-07-18:
```
BaseClass       SubClass        Protocol        Meaning
0xFF            0x..            0x..            Vendor specific
```

and...
```
12d1:1506 Huawei Technologies Co., Ltd. Modem/Networkcard
```

So it isn't actually CDC-ACM.
:-)

TDuxis, 2019-07-18:
Sorry, but this info has been there since the first time I posted the USB_desc?

I was of the impression that acm_terminal could be used as a serial terminal. I have understood others have used it this way. Someone used it for communication with a Huawei mobile phone. 
So why wouldn't it work with a modem? Whats the difference? Is this not a serial terminal?

I am asking here because I am looking for help.

xxxajk, 2019-07-18:
If it can be used as serial, then you need to know how to actually talk to it.
I've looked, and haven't found much information on it yet.

CDC-ACM is a specific specification that has a known API that is publicly open.

What pops up is a Vendor Specific interface, meaning that they made up something in order to be difficult morons, much like WinChipHead, FTDI, and others.

The trick is to find out what to send to it to do things.

xxxajk, 2019-07-18:
What is the actual model number for this?

TDuxis, 2019-07-18:
The modem is a Huawei E367u-2

I am still not getting any output from the printf_P,  what could be the cause for that?

xxxajk, 2019-07-18:
I'll look into that later today. First time I've heard of it not working.

TDuxis, 2019-07-18:
Great, thanks!

TDuxis, 2019-07-18:
Ok, so I have gotten this far. When I send AT, I get AT in return. I wonder, is this a local echo, or is it the Huawei actually returning the AT?

[image]

Here's the same thing with debug info

Waiting for Connection...

Configuring: parent = 0, port = 1, speed = 1

Configuring PktSize 0x40,  rcode: 0x00, retries 0,
ctrlReq2: left: 64, read:64, nbytes 64
RESULT: 0x00 0x00 0x12 0x0040000001000680
ctrlReq3: acceptBuffer sz 18 nbytes 64 left 46

0 retries.
DevDescr 2nd poll, bMaxPacketSize0:64
ctrlReq2: left: 18, read:18, nbytes 18
RESULT: 0x00 0x01 0x12 0x0012000001000680
0 retries.
configs: 1
ctrlReq2: left: 9, read:9, nbytes 9
RESULT: 0x00 0x01 0x09 0x0009000002000680
CONFIGURATION: 0, bNumInterfaces 2, wTotalLength 62
bLength: 9 bDescriptorType: 02
eating 8
bLength: 9 bDescriptorType: 04
INTERFACE DESCRIPTOR FOUND
Getting 3 endpoints
bLength: 7 bDescriptorType: 05
ENDPOINT DESCRIPTOR: 0
ENDPOINT DESCRIPTOR OK
eating 0
bLength: 7 bDescriptorType: 05
ENDPOINT DESCRIPTOR: 1
ENDPOINT DESCRIPTOR OK
eating 0
bLength: 7 bDescriptorType: 05
ENDPOINT DESCRIPTOR: 2
ENDPOINT DESCRIPTOR OK
eating 0
eating 1
ENDPOINT DESCRIPTORS FILLED
First endpoit class 0xff
TestInterface VID:12d1 PID:1506 Class:00 Subclass:00 Protocol 00
Interface data: Class:ff Subclass:01 Protocol 01, number of endpoints 3
Parent: 00, bAddress: 01
No driver at index 2
No driver at index 3
No driver at index 4
No driver at index 5
No driver at index 6
No driver at index 7
No driver at index 8
No driver at index 9
No driver at index 10
No driver at index 11
No driver at index 12
No driver at index 13
No driver at index 14
No driver at index 15
No driver for this interface.
bLength: 9 bDescriptorType: 04
INTERFACE DESCRIPTOR FOUND
Getting 2 endpoints
bLength: 7 bDescriptorType: 05
ENDPOINT DESCRIPTOR: 0
ENDPOINT DESCRIPTOR OK
eating 0
bLength: 7 bDescriptorType: 05
ENDPOINT DESCRIPTOR: 1
ENDPOINT DESCRIPTOR OK
eating 0
eating 1
ENDPOINT DESCRIPTORS FILLED
Second  endpoit class 0x0a
TestInterface VID:12d1 PID:1506 Class:00 Subclass:00 Protocol 00
Interface data: Class:0a Subclass:01 Protocol 01, number of endpoints 2
Parent: 00, bAddress: 01
Driver 1 supports this interface
Driver 1 can be used for this interface
USB_INTERFACE END OF STREAM
ctrlReq2: left: 9, read:9, nbytes 9
RESULT: 0x00 0x01 0x09 0x0009000002000680
CONFIGURATION: 1, bNumInterfaces 2, wTotalLength 62
Best configuration is 1, enumerating interfaces.
bLength: 9 bDescriptorType: 02
eating 8
bLength: 9 bDescriptorType: 04
INTERFACE DESCRIPTOR FOUND
Getting 3 endpoints
bLength: 7 bDescriptorType: 05
ENDPOINT DESCRIPTOR: 0
ENDPOINT DESCRIPTOR OK
eating 0
bLength: 7 bDescriptorType: 05
ENDPOINT DESCRIPTOR: 1
ENDPOINT DESCRIPTOR OK
eating 0
bLength: 7 bDescriptorType: 05
ENDPOINT DESCRIPTOR: 2
ENDPOINT DESCRIPTOR OK
eating 0
eating 1
ENDPOINT DESCRIPTORS FILLED
Second  endpoit class 0x0a
AttemptConfig: parent = 0, port = 1
Driver 1 supports this interface
Interface Configured
bLength: 9 bDescriptorType: 04
INTERFACE DESCRIPTOR FOUND
Getting 2 endpoints
bLength: 7 bDescriptorType: 05
ENDPOINT DESCRIPTOR: 0
ENDPOINT DESCRIPTOR OK
eating 0
bLength: 7 bDescriptorType: 05
ENDPOINT DESCRIPTOR: 1
ENDPOINT DESCRIPTOR OK
eating 0
eating 1
ENDPOINT DESCRIPTORS FILLED
Second  endpoit class 0x0a
AttemptConfig: parent = 0, port = 1
Driver 1 is already consumed @ 01
No driver at index 2
No driver at index 3
No driver at index 4
No driver at index 5
No driver at index 6
No driver at index 7
No driver at index 8
No driver at index 9
No driver at index 10
No driver at index 11
No driver at index 12
No driver at index 13
No driver at index 14
No driver at index 15
No interface driver for this interface.Driver 0 @ 00 Skipped
Driver 1 @ 01 Initialize

Connected.

Total endpoints = (4)4
Driver 2 no driver at this index.
Driver 3 no driver at this index.
Driver 4 no driver at this index.
Driver 5 no driver at this index.
Driver 6 no driver at this index.
Driver 7 no driver at this index.
Driver 8 no driver at this index.
Driver 9 no driver at this index.
Driver 10 no driver at this index.
Driver 11 no driver at this index.
Driver 12 no driver at this index.
Driver 13 no driver at this index.
Driver 14 no driver at this index.
Driver 15 no driver at this index.
outTransfer: addr: 0x01 ep: 0x01 nbytes: 0x0001 data: 0x0x21b2
outTransfer: SetAddress 0x00
outTransfer: addr: 0x01 ep: 0x01 nbytes: 0x0001 data: 0x0x21b2
outTransfer: SetAddress 0x00
AoutTransfer: addr: 0x01 ep: 0x01 nbytes: 0x0001 data: 0x0x21b2
outTransfer: SetAddress 0x00
T

TDuxis, 2019-07-18:
Ok, so the E367 has two configurations, and if I understand things correct, one of these two configurations must be selected by sending a SetConfiguration command. Is this handled by UHS3/acm_terminal? If so, how does it know which configuration to select?
Hence, I assume no. Then this is one thing I need to sort out to get this working, right?

TDuxis, 2019-07-18:
Hmmm, I am clearly onto something with the setting above. Because I have experimented with different setups of the modem, and the one above is the only one that gets "Connected", and returns the command.
Question is, why am I not seeing any response (other than the returned AT command"?

Any suggestions.

If it is of any help, the modem setting is   AT^SETPORT="FF;1,2"  where 1=Modem and 2=PCUI and from what I understand PCUI is the actual AT command interface.

TDuxis, 2019-07-18:
YES!!!!!

YES YES YES!!!!

I can see data coming from the device!!!! 

However, I am still not getting any response from my AT commands :(

So, conclusion right now, the endpoint for receiving data is correct, but not for sending. Odd though that what I send is echoed back (local echo?)

TDuxis, 2019-07-18:
Ok, so here's what the debug info looks like right now.

The ^BOOT:..... is what the USB device sends 

As can be seen, there's no response on my AT and ATI commands that I sent.

Waiting for Connection...

Configuring: parent = 0, port = 1, speed = 1

Configuring PktSize 0x40,  rcode: 0x00, retries 0,
ctrlReq2: left: 64, read:64, nbytes 64
RESULT: 0x00 0x00 0x12 0x0040000001000680
ctrlReq3: acceptBuffer sz 18 nbytes 64 left 46

0 retries.
DevDescr 2nd poll, bMaxPacketSize0:64
ctrlReq2: left: 18, read:18, nbytes 18
RESULT: 0x00 0x01 0x12 0x0012000001000680
0 retries.
configs: 1
ctrlReq2: left: 9, read:9, nbytes 9
RESULT: 0x00 0x01 0x09 0x0009000002000680
CONFIGURATION: 0, bNumInterfaces 2, wTotalLength 62
bLength: 9 bDescriptorType: 02
eating 8
bLength: 9 bDescriptorType: 04
INTERFACE DESCRIPTOR FOUND
Getting 3 endpoints
bLength: 7 bDescriptorType: 05
ENDPOINT DESCRIPTOR: 0
Endpoint Info 1: 81 
Endpoint Info 1: 3 
Endpoint Info 1: 40 
Endpoint Info 1: 0 
Endpoint Info 1: 5 
ENDPOINT SCRIPTOR OK
eating 0
bLength: 7 bDescriptorType: 05
ENDPOINT DESCRIPTOR: 1
Endpoint Info 1: 82 
Endpoint Info 1: 2 
Endpoint Info 1: 40 
Endpoint Info 1: 0 
Endpoint Info 1: 20 
ENDPOINT SCRIPTOR OK
eating 0
bLength: 7 bDescriptorType: 05
ENDPOINT DESCRIPTOR: 2
Endpoint Info 1: 1 
Endpoint Info 1: 2 
Endpoint Info 1: 40 
Endpoint Info 1: 0 
Endpoint Info 1: 20 
ENDPOINT SCRIPTOR OK
eating 0
eating 1
ENDPOINT DESCRIPTORS FILLED
First endpoit class 0xff
TestInterface VID:12d1 PID:1506 Class:00 Subclass:00 Protocol 00
Interface data: Class:ff Subclass:01 Protocol 01, number of endpoints 3
Parent: 00, bAddress: 01
No driver at index 2
No driver at index 3
No driver at index 4
No driver at index 5
No driver at index 6
No driver at index 7
No driver at index 8
No driver at index 9
No driver at index 10
No driver at index 11
No driver at index 12
No driver at index 13
No driver at index 14
No driver at index 15
No driver for this interface.
bLength: 9 bDescriptorType: 04
INTERFACE DESCRIPTOR FOUND
Getting 2 endpoints
bLength: 7 bDescriptorType: 05
ENDPOINT DESCRIPTOR: 0
Endpoint Info 1: 83 
Endpoint Info 1: 2 
Endpoint Info 1: 40 
Endpoint Info 1: 0 
Endpoint Info 1: 20 
ENDPOINT SCRIPTOR OK
eating 0
bLength: 7 bDescriptorType: 05
ENDPOINT DESCRIPTOR: 1
Endpoint Info 1: 2 
Endpoint Info 1: 2 
Endpoint Info 1: 40 
Endpoint Info 1: 0 
Endpoint Info 1: 20 
ENDPOINT SCRIPTOR OK
eating 0
eating 1
ENDPOINT DESCRIPTORS FILLED
Second endpoit class 0x a
TestInterface VID:12d1 PID:1506 Class:00 Subclass:00 Protocol 00
Interface data: Class:0a Subclass:01 Protocol 01, number of endpoints 2
Parent: 00, bAddress: 01
Driver 1 supports this interface
Driver 1 can be used for this interface
USB_INTERFACE END OF STREAM
ctrlReq2: left: 9, read:9, nbytes 9
RESULT: 0x00 0x01 0x09 0x0009000002000680
CONFIGURATION: 1, bNumInterfaces 2, wTotalLength 62
Best configuration is 1, enumerating interfaces.
bLength: 9 bDescriptorType: 02
eating 8
bLength: 9 bDescriptorType: 04
INTERFACE DESCRIPTOR FOUND
Getting 3 endpoints
bLength: 7 bDescriptorType: 05
ENDPOINT DESCRIPTOR: 0
Endpoint Info 1: 81 
Endpoint Info 1: 3 
Endpoint Info 1: 40 
Endpoint Info 1: 0 
Endpoint Info 1: 5 
ENDPOINT SCRIPTOR OK
eating 0
bLength: 7 bDescriptorType: 05
ENDPOINT DESCRIPTOR: 1
Endpoint Info 1: 82 
Endpoint Info 1: 2 
Endpoint Info 1: 40 
Endpoint Info 1: 0 
Endpoint Info 1: 20 
ENDPOINT SCRIPTOR OK
eating 0
bLength: 7 bDescriptorType: 05
ENDPOINT DESCRIPTOR: 2
Endpoint Info 1: 1 
Endpoint Info 1: 2 
Endpoint Info 1: 40 
Endpoint Info 1: 0 
Endpoint Info 1: 20 
ENDPOINT SCRIPTOR OK
eating 0
eating 1
ENDPOINT DESCRIPTORS FILLED
Second endpoit class 0x a
AttemptConfig: parent = 0, port = 1
Driver 1 supports this interface
Interface Configured
bLength: 9 bDescriptorType: 04
INTERFACE DESCRIPTOR FOUND
Getting 2 endpoints
bLength: 7 bDescriptorType: 05
ENDPOINT DESCRIPTOR: 0
Endpoint Info 1: 83 
Endpoint Info 1: 2 
Endpoint Info 1: 40 
Endpoint Info 1: 0 
Endpoint Info 1: 20 
ENDPOINT SCRIPTOR OK
eating 0
bLength: 7 bDescriptorType: 05
ENDPOINT DESCRIPTOR: 1
Endpoint Info 1: 2 
Endpoint Info 1: 2 
Endpoint Info 1: 40 
Endpoint Info 1: 0 
Endpoint Info 1: 20 
ENDPOINT SCRIPTOR OK
eating 0
eating 1
ENDPOINT DESCRIPTORS FILLED
Second endpoit class 0x a
AttemptConfig: parent = 0, port = 1
Driver 1 supports this interface
Interface Configured
Driver 0 @ 00 Skipped
Driver 1 @ 01 Initialize

Connected.

Total endpoints = (6)6
Driver 2 no driver at this index.
Driver 3 no driver at this index.
Driver 4 no driver at this index.
Driver 5 no driver at this index.
Driver 6 no driver at this index.
Driver 7 no driver at this index.
Driver 8 no driver at this index.
Driver 9 no driver at this index.
Driver 10 no driver at this index.
Driver 11 no driver at this index.
Driver 12 no driver at this index.
Driver 13 no driver at this index.
Driver 14 no driver at this index.
Driver 15 no driver at this index.

^BOOT:40354294,0,0,0,99
outTransfer: addr: 0x01 ep: 0x02 nbytes: 0x0001 data: 0x0x21b2
outTransfer: SetAddress 0x00
outTransfer: addr: 0x01 ep: 0x02 nbytes: 0x0001 data: 0x0x21b2
outTransfer: SetAddress 0x00
AoutTransfer: addr: 0x01 ep: 0x02 nbytes: 0x0001 data: 0x0x21b2
outTransfer: SetAddress 0x00
T
outTransfer: addr: 0x01 ep: 0x02 nbytes: 0x0001 data: 0x0x21b2
outTransfer: SetAddress 0x00
outTransfer: addr: 0x01 ep: 0x02 nbytes: 0x0001 data: 0x0x21b2
outTransfer: SetAddress 0x00
AoutTransfer: addr: 0x01 ep: 0x02 nbytes: 0x0001 data: 0x0x21b2
outTransfer: SetAddress 0x00
ToutTransfer: addr: 0x01 ep: 0x02 nbytes: 0x0001 data: 0x0x21b2
outTransfer: SetAddress 0x00
I

^BOOT:40354294,0,0,0,99

^BOOT:40354294,0,0,0,99

^BOOT:40354294,0,0,0,99

TDuxis, 2019-07-19:
Well, well, well.

Look at what popped up in the serial window

"COMMAND NOT SUPPORT"

This has to mean that something of what I am sending is coming through somewhere :)

TDuxis, 2019-07-19:
YES!!!! 
YES YES YES YES!!!

Look at what I got now, as a response

Manufacturer: huawei
Model: E367
Revision: 11.807.05.00.00
IMEI: 353613040354295
+GCAP: +CGSM,+DS,+ES

OK

:)
I think I got it.
Now it seems to work!
Next step is to try to send an sms, whish me luck :)

TDuxis, 2019-07-19:
HA!

SMS Sent and received!!!   

YIHAAAA! I DID IT! :D

My Arduino can now send SMS through a common Huawei USB modem! 

Mission accomplished :)

xxxajk, 2019-07-19:
This is awesome. Pull requests are welcome.
I'm assuming this integrates well?

xxxajk, 2019-07-19:
Also, this was probably easier to do in UHS3, than UHS2.
Note that in the future, at some point, I may be able to support the actual networking side, but this will require a better board. Imagine if you will being able to do more than just SMS, perhaps even voice...
All is indeed possible with a better MCU.

closed 2022-09-23 by xxxajk