felis/UHS30 #58: UHS_HID_KB_MOUSE.ino producing unrecognizable character

rtek1000, 2019-11-05:
> HID RAW/KEYBOARD/MOUSE demo Begin.
> BOOT_KEYBOARD
> BOOT_MOUSE
>  @⸮ @⸮HIDBOOT-RAW-MOUSE

After comparing with UHS_HID_RAW.ino, found:

Producing unrecognizable character:
`void onStart(UHS_HID_base *d) { printf_P("HID driver type %d started, Subclass %02x, Protocol %02x ",`

Producing normal character:
`void onStart(UHS_HID_base *d) { printf_P(PSTR("HID driver type %d started, Subclass %02x, Protocol %02x\r\n"),`

rtek1000, 2019-11-05:
After the above modification, other characters started playing but also unrecognizable:

> HID RAW/KEYBOARD/MOUSE demo Begin.
> BOOT_KEYBOARD
> BOOT_MOUSE
> HID driver type 2 started, Subclass 01, Protocol 02
> HIDBOOT-RAW-MOUSE
>    @ @⸮ @⸮⸮@ ⸮⸮@  @⸮ @⸮   @ @⸮ @⸮⸮@ ⸮⸮@  @⸮ @⸮

```C++
case UHS_HID_mouse:
          printf_P(PSTR("Mouse buttons left %s right %s mid %s fourth %s fifth %s motion (X,Y) %4d,%4d wheel %4d"), squeek->bmLeftButton == 1 ? "t" : "f", squeek->bmRightButton == 1 ? "t" : "f", squeek->bmMiddleButton == 1 ? "t" : "f", squeek->bmButton4 == 1 ? "t" : "f", squeek->bmButton5 == 1 ? "t" : "f", squeek->dX, squeek->dY, squeek->wheel1);
```
If line printf_P is commented out unrecognizable characters stop appearing, Something may be wrong in this line.

xxxajk, 2019-11-06:
Debug mode uses a lot of RAM. The problem is that you are out of RAM.

xxxajk, 2019-11-06:
And yes, should be PSTR

rtek1000, 2019-11-18:
I tried using the UHS_HID_RAW.ino example with Mega2560, and noticed that a message is displayed even when disabling debug mode, I may have to add some #ifdef entry there:

UHS_HID.h file:

Line 146:

From:
`printf("Length %d ", length);`
To:
`//printf("Length %d ", length);`

xxxajk, 2019-11-20:
Nice find, I'll patch it when I get the chance.