felis/UHS30 #61: UHS_MIDI: Arturia Beatstep Pro not working

recliq, 2020-05-03:
Hi,

I'm trying to use UHS_MIDI to connect an Arturia Beatstep Pro which is not working.
It seems to find and connect the device, but as soon as any data from the device arrives it just disconnects the device...

``` USB MIDI Converter example.

Configuring: parent = 0, port = 1, speed = 1

Configuring PktSize 0x40,  rcode: 0x00, retries 0
ep entry for interface 2 ep 0 max packet size = 64
ctrlReq2: left: 64, read:64, nbytes 64
RESULT: 0x00 0x00 0x12 0x0040000001000680
ctrlReq3: acceptBuffer sz 18 nbytes 64 left 46

ep entry for interface 2 ep 0 max packet size = 64
0 retries.
DevDescr 2nd poll, bMaxPacketSize0:64
ep entry for interface 11 ep 0 max packet size = 64
ctrlReq2: left: 18, read:18, nbytes 18
RESULT: 0x00 0x01 0x12 0x0012000001000680
ep entry for interface 2 ep 0 max packet size = 64
0 retries.
configs: 1
ep entry for interface 2 ep 0 max packet size = 64
ctrlReq2: left: 9, read:9, nbytes 9
RESULT: 0x00 0x01 0x09 0x0009000002000680
CONFIGURATION: 0, bNumInterfaces 2, wTotalLength 133
ep entry for interface 2 ep 0 max packet size = 64
bLength: 9 bDescriptorType: 02
eating 8
bLength: 9 bDescriptorType: 04
INTERFACE DESCRIPTOR FOUND
Getting 0 endpoints
eating 1
ENDPOINT DESCRIPTORS FILLED
TestInterface VID:1c75 PID:0287 Class:00 Subclass:00 Protocol 00
Interface data: Class:01 Subclass:01 Protocol 01, number of endpoints 0
Parent: 00, bAddress: 01
MIDI: checking vid 1c75, pid 0287, iface class 1, subclass 1
No driver at index 1
No driver at index 2
No driver at index 3
No driver at index 4
No driver at index 5
No driver at index 6
No driver at index 7
No driver at index 8
No driver at index 9
No driver at index 10
No driver at index 11
No driver at index 12
No driver at index 13
No driver at index 14
No driver at index 15
No driver for this interface.
bLength: 9 bDescriptorType: 24
eating 8
bLength: 9 bDescriptorType: 04
INTERFACE DESCRIPTOR FOUND
Getting 2 endpoints
bLength: 7 bDescriptorType: 24
eating 5
bLength: 6 bDescriptorType: 24
eating 4
bLength: 6 bDescriptorType: 24
eating 4
bLength: 6 bDescriptorType: 24
eating 4
bLength: 6 bDescriptorType: 24
[... 286 lines cut ...]
          baAssocJackID( 1)       5
      Endpoint Descriptor:
        bLength                 9
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
        bRefresh                0
        bSynchAddress           0
        MIDIStreaming Endpoint Descriptor:
          bLength                 6
          bDescriptorType        37
          bDescriptorSubtype      1 (GENERAL)
          bNumEmbMIDIJack         2
          baAssocJackID( 0)       3
          baAssocJackID( 1)       7
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled
```
The Beatstep Pro reports 2 Midi-In and MidiOut Ports, one seems to be used by for the configuration utility only...

WonkoSan, 2020-05-11:
Interesting issue.  I am also working on a Midi project but I am going to wire up a 5-pin 5 connector to use as my Midi out port instead of Midi-over-USB.

xxxajk, 2020-05-11:
The biggest issue with this brand is that they expect USB 2 HIGH speed, and they don't queue keys/pads/etc. WHen this happens, either the atmega gets overwhelmed and will miss key/pad presses, or disconnect. I've seen both happen. Doesn't happen on the Yamaha keyboard or the other one I have here.

Since the atmega just can't keep up,  I recommend getting a Teensy 3.6 from PJRC.com. There's no need to add a host shield (host is built in) and you get high-speed USB instead of just low and full speed. On top of that, you can use huge buffers to store key and pad smashing and keep up with the data if you have to. Paul's support is second to none too.

recliq, 2020-05-11:
Thank you for your detailed answere @xxxajk . I will try a teensy and see if that works.

closed 2020-05-11 by recliq