felis/UHS30 #64: Error compiling UHS_HID_KB_MOUSE.ino

tinkerBOY-git, 2020-10-19:
Trying to compile the example UHS_HID_KB_MOUSE.ino but keeps getting this error:

```
c:/users/gio/appdata/local/arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: warning: changing start of section .bss by 240 bytes

c:/users/gio/appdata/local/arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: warning: changing start of section .bss by 240 bytes

c:/users/gio/appdata/local/arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: warning: changing start of section .bss by 240 bytes

c:/users/gio/appdata/local/arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: warning: changing start of section .bss by 240 bytes

c:/users/gio/appdata/local/arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: warning: changing start of section .bss by 240 bytes

C:\Users\gio\AppData\Local\Temp\arduino_build_132947\libraries\ISR_safe_memory\mlock.c.o: In function `__malloc_lock':

D:\Documents\Arduino\libraries\ISR_safe_memory/mlock.c:153: undefined reference to `__retarget_lock_acquire_recursive'

D:\Documents\Arduino\libraries\ISR_safe_memory/mlock.c:155: undefined reference to `__lock___malloc_lock_object'

C:\Users\gio\AppData\Local\Temp\arduino_build_132947\libraries\ISR_safe_memory\mlock.c.o: In function `__malloc_unlock':

D:\Documents\Arduino\libraries\ISR_safe_memory/mlock.c:162: undefined reference to `__retarget_lock_release_recursive'

D:\Documents\Arduino\libraries\ISR_safe_memory/mlock.c:177: undefined reference to `__lock___malloc_lock_object'

collect2.exe: error: ld returned 1 exit status
```

Any idea?

xxxajk, 2020-10-19:
Please tell me the following:
1: Which Arduino board are you using
2: Which USB board or native USB

tinkerBOY-git, 2020-10-20:
> Please tell me the following:
> 1: Which Arduino board are you using
> 2: Which USB board or native USB

I'm compiling for:
1. Arduino Zero
2. Native USB port

xxxajk, 2020-10-20:
The Zero's native USB port isn't supported, unless you are writing the port for it.
As far as the compiler errors, this is because it's either not using newlib, or a broken variant of newlib.

tinkerBOY-git, 2021-02-18:
Sorry I wasn't so specific with my inquiry. I was actually compiling for ATSAMD21G18A which is the chip inside the Arduino Zero board. How do you install or get the new newlib?

xxxajk, 2021-02-22:
Newlib is part of arduino. Unless they horribly changed something, it should be available.

closed 2022-09-23 by xxxajk