felis/UHS30 #66: USB Host Shield Failed to Connect

Ynot-Dc, 2022-12-09:
I failed to build a connection between USB host Shield and My Device. I'm using USB host library 2.0 and USH30 in that i have used CDC_ACM as well which is available on GitHub.
Could You please help me out I am totally new in this.
 
My device specs are listed below:
VIRTUAL COM PORT
Device Type: Ports (COM & LPT)
Manufacturer: FTDI
Location: on USB Serial Converter 
Baud Rate:115200 baud* Format: 8 data bits, no parity, 1 stop bit*
Device Description : USB Serial Port 
USB Device Hardware Ids: FTDIBUS\COMPORT&VID_0403&PID_6015  

When I upload USBFTDILoopback.ino It won't give any response just print "Start", Please refer below screenshot
[image]

And in acm_terminal.ino it prints in Serial Monitor "waiting for connection....", Please refer below screenshot
[image]

xxxajk, 2022-12-12:
First, for `USBFTDILoopback.ino` you need to connect the FTDI RX and TX together, then you will see the data looping back.

UHS3's demo is different, but can be made to do the same, however it involves your interaction. i.e. entering some text in the message box, and sending it. The text you entered will be sent back to the screen.

Important question:
What board? MEGA2560? Teensy? Something else?

Ynot-Dc, 2022-12-14:
The board I am using is Arduino UNO R3.

I have connected the USB Host Shield to Arduino UNO and with USB Host Shield I have connected My Device, 

when I connect My Device directly to the System it works but when I connect to USB Host Shield it won't give any response. 
I have also soldered the USB Host Shield for Power.

xxxajk, 2022-12-16:
UHS3 might not work well on the UNO (lack of enough ram) a mega2560 is pretty much the lowest end board that works with it, without doing all sorts of nasty tricks. My guess for UHS3 is possibly not enough memory. If you have a Due available, that should work pretty well, but I have not tested that in a very long time.