felis/UHS30 #68: Compatibility with M5Stack USB v1.2 Module

jaylfc, 2024-10-07:
Hi and great work!

I'm just wondering if this library would work with the m5stack usb host module?

https://docs.m5stack.com/en/module/USB%20v1.2%20Module

Thanks in advance,
Jay.

xxxajk, 2024-10-08:
Yes, it should be able to do so.

jaylfc, 2024-10-08:
Awesome, what should I put in my main ino file to override the pin configs? Cheers!

xxxajk, 2024-10-09:
Depends, what MCU is being used?

jaylfc, 2024-10-09:
I intend to use it with the M5Stack CoreS3, so esp32-s3.

Cheers!

xxxajk, 2024-10-10:
In order for that to work, you will need to do 2 things:

1: You need to make patches that make the heap thread safe if it is not already.
https://github.com/felis/UHS30/tree/master/libraries/ISR_safe_memory
as far as I know the esp32's heap should be isr safe? I don't really remember if it is.

2: You need to write a class-based isr handler that either allows reentrant IRQ or simulates it by using a software IRQ.
That code gets added into the two files here:
https://github.com/felis/UHS30/tree/master/libraries/dyn_SWI

jaylfc, 2024-10-10:
Thanks for that, not too sure how to do that but I'll try and figure it out. Thanks again :)

xxxajk, 2024-10-11:
Shouldn't be too difficult, you just need to understand the MCU and libc you are working with, and you also have ARM and AVR code to draw upon.

jaylfc, 2024-10-13:
I've just never approached something like this before, but I'll give it a go. Might have to see if I can use a little ai help ha! Thanks guys, really appreciate it!

xxxajk, 2024-10-17:
AI isn't going to be much help, and an LLM isn't actually AI anyway. 
Never seen it puke out anything usable.

That said, you do have how it works on other processors, read the code, get the datasheet for the MCU you are using, and it shouldn't be all that difficult to get what you want.